commaai/panda #2: Single-wire GMLAN support

vntarasov, 2017-05-08:
I don't have access to create a pull request.

What the attached diff does:
-  Fixes off-by-one error in can bus speed calculation, that was preventing single-wire GMLAN from working on my Chevy Volt gen 2.
- Makes it easier to understand, debug, and change the can speed configuration code.
- Enables sending/receiving extended addresses (majority of messages on single-wire GMLAN on my car).

[panda.patch.txt](https://github.com/commaai/panda/files/982669/panda.patch.txt)

geohot, 2017-05-09:
Hmm, what do you mean you don't have access? It should be public. Can you screenshot the page showing no access?

geohot, 2017-05-09:
Oh not sure what you were trying, but I don't see a panda repo in your account.

In order to do a pull request, first fork the repo to your account. Then push your changes to the fork. Then click pull request. It should work.

vntarasov, 2017-05-09:
Yep, fork is what I missed. I talked to @edsammy earlier and he helped me realize that since I don't have push rights, I shouldn't try creating branches in the Panda repo directly, but fork it first, and create pull request from there. Will do that now.

commit 37a87ff2 (2019-10-25): Linter fix #2
commit a12a148d (2019-10-25): Uno (#274)

* Added uno

* Added usb switch support

* Added PWM and IR power functions

* Implemented bootkick

* Added uno as a new hw type

* Bumped version

* Added fan control and tach readout

* WIP: RTC support

* Working RTC

* Fixed python

* Misra compliance

* Added USB control messages for fan/IR power

* Added USB commands + tests for fan & IR control. Fixed bootstub and pedal compilation

* Added IR and fan to power saving mode

* Changed defaults

* Fix safety considering uno

* passing safety now

* Minor UNO tweaks

* Fixed version

* More minor temporary tweaks

* Removed usb [... cut]
closed 2017-05-09 by vntarasov