commaai/panda #4: Panda firmware fails to compile with arm-none-eabi-gcc 5.4.1 20160919 (default std=gnu11)

diamondman, 2017-06-09:
Compiling the panda firmware on a Debian Stretch machine (using the debian provided gcc-arm-none-eabi version 5.4.1) fails complaining about inline functions not being available at link time. For example:
`undefined reference to 'early'`.

This stems from the C standard being used by GCC changing from `gnu90` to `gnu11` between gcc version 4.9 to 5.1 as detailed in the accepted post [here](https://stackoverflow.com/questions/14737104/what-is-the-default-c-mode-for-the-current-gcc-especially-on-ubuntu). In short, the semantics around the `inline` keyword has [changed between standards](https://stackoverflow.com/questions/16245521/c99-inline-function-in-c-file/16245669#16245669).

It appears that the original developer wrote the firmware using ubuntu's Xenial gcc-arm-none-eabi package which is currently on [version 4.9.3](https://packages.ubuntu.com/xenial/gcc-arm-none-eabi). Since Ubuntu will inevitably update to the [5.x](https://packages.debian.org/stretch/gcc-arm-none-eabi) line sometime in the future, we should fix this now.

The easiest solution is to explicitly mark the std type as  `gnu90`, thus restoring the default behavior of the older GCC version. No matter what is done, I believe that a C standard should be selected for the project to prevent these types of issues in the future.

But this issue raises a question: why are these functions inline?
The compiler is free to ignore the inline directive, as well as choose to inline a function that is not explicitly marked to be inlined [[1]](https://stackoverflow.com/a/1626269/627525). Furthermore, the firmware is compiled with -O0, and inlining is only done on optimizing compilations: "Inlining of functions is an optimization and it really “works” only in optimizing compilation. If you don't use -O, no function is really inline." [[gcc manual]](https://gcc.gnu.org/onlinedocs/gcc-3.3.6/gcc/Inline.html).

I argue that most of these inline functions should be stripped of their inline qualifier and split into a header file and a source file, as inlining can still be done across compilation units. This would also allow us to use more modern C semantics in the future. 

I am happy to make a PR with either of the proposed changes. What is the preferred way to resolve this issue?

geohot, 2017-06-10:
Feel free to remove inline everywhere. You are right that it's an optimization. Re: source files and header files, feel free to change as well if you desire. Comes from a time when the build system was less mature, and to speed up compilation since the files are so small.

diamondman, 2017-06-14:
A PR exists that fixes this issue.

PR #6 (2017-06-14): Remove inline keywords, cleaned up build.md, set compiler standard.
Closes #4 
Removed inline keywords on all functions. 
Split early.h into a header and c file.
Updated Makefile rules to be more consistent.
closed 2017-06-14 by geohot