commaai/panda #8: Panda python api can try to send CAN messages over 8 bytes and garbles the result.

diamondman, 2017-06-14:
Can is only capable of sending datapackets up to 8 bytes long. There is no limit on the message size that the python code will try to send. The results are messed up data.

**Correct Behavior**
```
In [22]: from panda import Panda;p = Panda(Panda.list()[0])
opening device .....

In [23]: p.can_send(0x13, b'12345678', 1)
DAT: b'\x01\x00`\x02\x18\x00\x00\x0012345678'

In [24]: p.can_recv()
Out[24]: 
[(19, 47928, bytearray(b'12345678'), 3),
 (19, 47928, bytearray(b'12345678'), 1)]
```

**Too long message**
```
In [22]: from panda import Panda;p = Panda(Panda.list()[0])
opening device .....

In [25]: p.can_send(0x13, b'123456789', 1)
DAT: b'\x01\x00`\x02\x19\x00\x00\x00123456789'

In [26]: p.can_recv()
Out[26]: 
[(448, 25113, bytearray(b''), 2),
 (448, 25113, bytearray(b''), 0),
 (19, 25175, bytearray(b'12345678'), 3),
 (19, 25175, bytearray(b'12345678'), 1)]
```

linagee, 2017-06-14:
Only 8 bytes assuming you are talking about regular CAN and not https://en.wikipedia.org/wiki/ISO_15765-2
(GMLAN uses this, not sure how often they are sending these longer messages.)
This is still 8 bytes per frame, just split into different messages.

geohot, 2017-06-20:
We should add an assert to the library.

closed 2017-06-26 by geohot