commaai/panda #28: kline_recv not working

quillford, 2017-07-31:
not sure if k line is fully implemented yet, but it just hangs when panda.kline_recv() is called. i'm communicating over wifi and panda.can_recv() works fine. panda is fully up to date according to ios chffr

geohot, 2017-07-31:
It should work. Are you sending something on kline? Unlike CAN it blocks.

quillford, 2017-07-31:
Not trying to send anything, just trying to read. I also tried over usb but the driver isn't working.

edsammy, 2017-08-25:
@quillford is this still an issue for you?

quillford, 2017-08-25:
not sure. one of the socket headers broke off when I was trying to remove my panda from the stocket, so I can't test until I fix that. the height on those headers seems odd, do you have a part number for it?

edsammy, 2017-08-26:
Its these low profile ones from adafruit
https://www.adafruit.com/product/3008

closed 2017-08-29 by geohot