commaai/panda #33: the shipping order of panda

sgong4, 2017-08-18:
hi:
    i have placed an order for a panda in  2017/7/24 , until now i don't receive any Logistics information about it.
    i have send emails to george@comma.ai. but no one answer me .
    Is it because we are in China, leading us to cann't receive logistics information, or other problems caused .   
   i really want a panda, can you help me . thank you very much.@geohot @joshenders 
   my order number is  (#102817092)

palmerit, 2017-08-21:
@sgong4 Pandas have not started shipping yet.  This was a pre-order as mentioned on the website when you ordered it.  The comma team (who I am not affiliated with) have stated they will announce via twitter when shipping starts.   Please be patient!

edsammy, 2017-08-25:
shipping will be begin shortly

your patience is appreciated!

closed 2017-08-25 by edsammy