commaai/panda #39: error when using pandacan flash

palmerit, 2017-08-29:
# Problem:
I've been getting a `LIBUSB_ERROR_IO` during "step 1" when doing the flash via pandacan.

# Steps to reproduce:
`sudo python -c "import panda; panda.flash_release()"`

# Result
```
Using panda with serial 410034000651363038363036
Fetching latest firmware from github.com/commaai/panda-artifacts
Fetching firmware from https://github.com/commaai/panda-artifacts/raw/master/panda-v0.9.7-ac6e0e67-RELEASE.zip
File Name                                             Modified             Size
bootstub.panda.bin                             2017-08-29 11:36:16        12348
panda.bin                                      2017-08-29 11:36:16        18516
user1.bin                                      2017-08-29 11:36:18       246980
user2.bin                                      2017-08-29 11:36:18       246980
boot_v1.5.bin                                  2016-05-20 16:53:08         3232
version                                        2017-08-29 11:36:18           23
0. Preparing to flash v0.9.7-ac6e0e67-RELEASE
1. Entering DFU mode
opening device 410034000651363038363036 0xddcc
connected
LIBUSB_ERROR_IO [-1]
2. Programming bootstub
programming 0 with length 2048
programming 1 with length 2048
programming 2 with length 2048
programming 3 with length 2048
programming 4 with length 2048
programming 5 with length 2048
programming 6 with length 2048
3. Flashing main code
opening device 410034000651363038363036 0xddee
connected
flash: version is v0.9.7-ac6e0e67-RELEASE
flash: unlocking
flash: erasing
flash: flashing
flash: resetting
opening device 410034000651363038363036 0xddcc
connected
4. Flashing ESP (slow!)
opening device 410034000651363038363036 0xddcc
connected
set baud to 230400
Connecting...
Connection timeout.
Running Cesanta flasher stub...
set baud to 230400
Writing 4096 @ 0x0... 4096 (100 %)
Writing 249856 @ 0x1000... 249856 (100 %)
Writing 249856 @ 0x81000... 249856 (100 %)
Writing 4096 @ 0x3fe000... 
5. Verifying version
opening device 410034000651363038363036 0xddcc
connected
dongle id: 4c537e87061caf72
v0.9.7-ac6e0e67-RELEASE should be v0.9.7-ac6e0e67-RELEASE
6. Success!
```

gameshack, 2017-08-29:
**Problem:**

I've been getting a LIBUSB_ERROR_PIPE during "step 1" when doing the flash via pandacan.

**Steps to reproduce:**

sudo pip install --upgrade pandacan && PYTHONPATH="" sudo python -c "import panda; panda.flash_release()"

**Result:**

Using panda with serial 000012800f51363038363036
Fetching latest firmware from github.com/commaai/panda-artifacts
Fetching firmware from https://github.com/commaai/panda-artifacts/raw/master/panda-v0.9.7-ac6e0e67-RELEASE.zip
File Name                                             Modified             Size
bootstub.panda.bin                             2017-08-29 11:36:16        12348
panda.bin                                      2017-08-29 11:36:16        18516
user1.bin                                      2017-08-29 11:36:18       246980
user2.bin                                      2017-08-29 11:36:18       246980
boot_v1.5.bin                                  2016-05-20 16:53:08         3232
version                                        2017-08-29 11:36:18           23
0. Preparing to flash v0.9.7-ac6e0e67-RELEASE
1. Entering DFU mode
opening device 000012800f51363038363036 0xddcc
connected
LIBUSB_ERROR_PIPE [-9]
2. Programming bootstub
programming 0 with length 2048
programming 1 with length 2048
programming 2 with length 2048
programming 3 with length 2048
programming 4 with length 2048
programming 5 with length 2048
programming 6 with length 2048
3. Flashing main code
opening device 000012800f51363038363036 0xddee
connected
flash: version is v0.9.7-ac6e0e67-RELEASE
flash: unlocking
flash: erasing
flash: flashing
flash: resetting
opening device 000012800f51363038363036 0xddcc
connected
4. Flashing ESP (slow!)
opening device 000012800f51363038363036 0xddcc
connected
set baud to 230400
Connecting...
Connection timeout.
Running Cesanta flasher stub...
set baud to 230400
Writing 4096 @ 0x0... 4096 (100 %)
Writing 249856 @ 0x1000... 249856 (100 %)
Writing 249856 @ 0x81000... 249856 (100 %)
Writing 4096 @ 0x3fe000... 
5. Verifying version
opening device 000012800f51363038363036 0xddcc
connected
dongle id: eda05d464a51ae49
v0.9.7-ac6e0e67-RELEASE should be v0.9.7-ac6e0e67-RELEASE
6. Success!

geohot, 2017-08-29:
Those errors don't matter. If it says success it succeeded. They happen because the USB "disappears" on reset.

Would accept a pull request to hide them, but Success means Success.

closed 2017-08-29 by geohot