commaai/panda #44: Cabana looping curiously

bbidault2, 2017-09-26:
I am seeing a curious behavior with cabana. When I click _launch realtime streaming_ then pairs and connect to the panda, it just brings me back to the same _welcome to cabana_ pop up window, same if I click _log in to view recorded drive_.

[image]

The panda blinks purple during the test (steady blue light and red light blinking) which I believe is the expected behavior when reading CAN data, it could be useful to update/extend the leds info on the wiki page. The panda is connected to the computer through USB.

This problem is seen on 4 different machines, two running Linux Ubuntu and two running Windows 7. The problem is both seen on Chrome and Chromium. 

The panda is up to date according to Chffr. Chffr works _okaishly_, it receives messages from the panda and the blue led of the wifi board blinks, although the app has a very hard time uploading the recorded drive. As read from the slack discussions, it seems like there is a trick with plane and wifi modes, it could be interesting to add a paragraph about it to the README and/or wiki. I am using an LG G3.

I have probably missed a part that needs to be accomplished prior to using the panda with cabana. Although with the pandacan library installed and the panda up to date, I don't know what it could be, I guess it's mostly a cabana issue. Also, neither socketcan nor _ifconfig_ detect the device as a CAN device.
I have already posted this issue in the cabana slack channel but, it's a slack channel, not the best place to report an issue.

Thank you for your work.

lemassykoi, 2017-10-27:
did you install the linux driver ?

bbidault2, 2017-10-30:
Thank you for your reply @lemassykoi . Yes, the linux driver is installed. After re-testing, I can tell the problem is still present. Do you use Chrome or Chromium for Cabana on Linux ?

lemassykoi, 2017-10-30:
I use Chrome and Ubuntu 16.04. but now I'm using Windows 10 x64, and chrome

lemassykoi, 2017-10-30:
Did you correctly ensble experimental web features and WebUSB, in chrome ?

bbidault2, 2017-10-30:
Yes, both features are enabled.

geohot, 2017-10-31:
So for live cabana, you don't want the Linux kernel driver installed, because it will put the panda in a different mode. Just the udev rules. cabana talks to the panda over userspace USB.

lemassykoi, 2017-11-01:
if you want to use socketcan, I think you might need the driver.
I didn't know about cabana which doesn't need it.
So, @bbidault2 did you put the 11-panda.rules file in /etc/udev/rules.d ?
`SUBSYSTEMS=="usb", ATTR{idVendor}=="bbaa", ATTR{idProduct}=="ddee", MODE:="0666"
SUBSYSTEMS=="usb", ATTR{idVendor}=="bbaa", ATTR{idProduct}=="ddcc", MODE:="0666"`
which is your linux ?

bbidault2, 2017-11-01:
The udev rules ! I can now use realtime streaming but it is still impossible to look view recorded drives or load an existing dbc file. Although, being able to use realtime streaming solves most of my problem. Thank you. 
I use linux Ubuntu 16.04.

lemassykoi, 2017-11-01:
so, your issue is solved ? ^^

bbidault2, 2017-11-01:
Yes that should do it, although I still don't understand why I can't load a previously recorded dbc file. Thanks again.

closed 2017-11-01 by bbidault2

lemassykoi, 2017-11-01:
why can't you ? how do you try ? by pasting it ? by github ?

bbidault2, 2017-11-02:
Either by clickint on _"Log in to View Recorded Drives"_ (see my original post), the app just loops back to that same popup window ; or by using _"load DBC"_, I can select a DBC file but then it just keeps loading.

lemassykoi, 2017-11-02:
You need to be logged as your G+ account, and to github if you want to load from it