commaai/panda #47: cannot connect to car

tspannhw, 2017-10-06:
I have the panda hooked up and it's lighting up.  it's plugged in with usb wire to my mac with panda installed and 1 of 2 things happens:

➜  panda git:(master) ✗ python3 examples/query_vin_and_stats.py
Traceback (most recent call last):
  File "examples/query_vin_and_stats.py", line 4, in <module>
    from panda import Panda
  File "/usr/local/lib/python3.6/site-packages/panda/__init__.py", line 11, in <module>
    from dfu import PandaDFU
ModuleNotFoundError: No module named 'dfu'

➜  examples git:(master) ✗ python query_vin_and_stats.py
Traceback (most recent call last):
  File "query_vin_and_stats.py", line 31, in <module>
    panda = Panda()
  File "/usr/local/lib/python2.7/site-packages/panda/__init__.py", line 116, in __init__
    self.connect(claim)
  File "/usr/local/lib/python2.7/site-packages/panda/__init__.py", line 159, in connect
    assert(self._handle != None)
AssertionError

Souryadeep, 2018-01-19:
examples git:(master) ✗ python query_vin_and_stats.py
Traceback (most recent call last):
File "query_vin_and_stats.py", line 31, in
panda = Panda()
File "/usr/local/lib/python2.7/site-packages/panda/init.py", line 116, in init
self.connect(claim)
File "/usr/local/lib/python2.7/site-packages/panda/init.py", line 159, in connect
assert(self._handle != None)
AssertionError

**This means your panda is not connected. Try connecting again.**

dieterbauwens, 2018-03-05:
Hi, I'm not a Python expert, so I cannot tell you how to structurally solve it, but the module python/dfu.py (which sits in your site-packages panda folder after you pip install pandacan) just needs to be in the working directory from where you run your code.

My context: Windows 7 64bit, Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:25:58)

closed 2019-11-13 by rbiasini