commaai/panda #52: finish panda linux kernel driver (please)

lemassykoi, 2017-10-26:
The linux kernel driver does not support changing the bitrate of the can bus interfaces. 
Changing the bitrate is required for this driver to be useful. 

When using the Caribou software (which uses the socket can interfaces provided by this driver) to communicate with my car, Caribou reported it failed to initialize a connection, and the car started to display errors. Since the errors vanished after removing the panda, they are likely due to the inability to change the bitrate through the driver, causing the panda to stay active at the wrong bitrate.

After normal operation returned, my car has artifacting on its dashboard display. I will update this post if it goes away after disconnecting the car battery for 5 minutes.

Souryadeep, 2017-12-14:
E: Unable to locate package linux-headers-$
E: Couldn't find any package by regex 'linux-headers-$'

When I run **apt-get install dkms gcc linux-headers-$ 4.4.0-103-generic make sudo**, I get the two above lines. Not sure why. Looking forward to advice.

bertramt, 2017-12-14:
@Souryadeep 
try: apt-get install dkms gcc linux-headers-$(uname -r) make sudo

If you have further problems visit slack or the forum to help you troubleshoot as your problem is unrelated to this issue.

closed 2019-11-13 by rbiasini