commaai/panda #56: can't reliably sending first message for reading VIN

gregjhogan, 2017-11-02:
I am trying to read the VIN from my 2017 CR-V and after connecting to the panda over USB the first message usually receives no response.  If I repeat sending the message it seems to reliably succeed on subsequent calls.

Here is how I can reliably reproduce this with my car:
```python
#!/usr/bin/env python
import time
import struct
from panda import Panda
from isotp import isotp_send, isotp_recv

if __name__ == "__main__":
  panda = Panda()
  panda.set_safety_mode(Panda.SAFETY_ELM327)

  panda.can_clear(0)
  time.sleep(0.1)

  for z in range(0,3):
    addr = 0x18DA10F1
    msg = "\x02\x09\x02".ljust(8, "\x00")
    print "S:", format(addr,'x'), msg.encode("hex")
    panda.can_send(addr, msg, 0)
    more = False
    for x in range(0, 100):
      msgs = panda.can_recv()
      for ids, ts, dat, bus in msgs:
        if bus == 0 and ids == ((addr >> 16 << 16) | (addr << 8 & 0xFFFF) | (addr >> 8 & 0xFF)):
          print "R:", format(ids,'x'), ("".join("%02x" % i for i in dat))
          print dat[5:]
          msg = "\x30\x00\x00".ljust(8, "\x00")
          panda.can_send(addr, msg, 0)
          print "S:", format(addr,'x'), msg.encode("hex")
          more = True
          break

      if more:
        break

    if more:
      for x in range(0, 100):
        msgs = panda.can_recv()
        for ids, ts, dat, bus in msgs:
          #if bus == 0 and ids == ((addr >> 16 << 16) | (addr << 8 & 0xFFFF) | (addr >> 8 & 0xFF)):
          if bus == 0 and ids>>16 == addr>>16:
            print "R:", format(ids,'x'), ("".join("%02x" % i for i in dat))
            print dat[1:]
    else:
      print "failed!"

  print "done"
```

here is the output (notice that the first iteration of the outer loop says failed! and the next two iterations successfully read the entire VIN):
```
$ sudo python query_vin_and_stats.py
opening device 4d003b00xxxxxxxxxxxxxxxx 0xddcc
connected
S: 18da10f1 0209020000000000
failed!

S: 18da10f1 0209020000000000
R: 18daf110 1014490201354a36
5J6
S: 18da10f1 3000000000000000
R: 18daf110 2152573248393248
RW2H92H
R: 18daf110 22xxxxxxxxxxxxxx
XXXXXXX

S: 18da10f1 0209020000000000
R: 18daf110 1014490201354a36
5J6
S: 18da10f1 3000000000000000
R: 18daf110 2152573248393248
RW2H92H
R: 18daf110 22xxxxxxxxxxxxxx
XXXXXXX

done
```

Souryadeep, 2017-12-13:
Can the isotp library be used in python2?

energee, 2017-12-13:
yes!

gregjhogan, 2017-12-13:
Could this be my issue?

https://github.com/commaai/panda/blob/master/drivers/windows/panda/panda.cpp#L156-L165

It is the windows driver, but it says it happens on linux, too.

Souryadeep, 2017-12-14:
from isotp import isotp_send, isotp_recv
  File "/home/ubuntu/python-can-isotp/isotp/__init__.py", line 9, in <module>
    raise NotImplementedError("Your version of Python does not offer support for CAN ISO-TP protocol. Support have been added since Python 3.7 on Linux build > 2.6.15.")
NotImplementedError: Your version of Python does not offer support for CAN ISO-TP protocol. Support have been added since Python 3.7 on Linux build > 2.6.15.

**This is what I am getting**

gregjhogan, 2018-02-12:
figured out what my issue was a while ago ... I needed to have something reading messages before I started writing messages.  Without something reading messages the RX buffer on the panda would get full and once it is full messages get dropped (so the first response to requesting the VIN was getting dropped).

closed 2018-02-12 by gregjhogan