commaai/panda #72: where can I find the wifi password of panda?

busyrock, 2017-12-01:
I can scan the wifi of panda when I connect panda with OBD, but this wifi need a password, where can I find this password? The wifi name is panda-a62e6c57f5a2a711 and I hope for some help. Thank you very much!

ratsbane, 2017-12-01:
On the chffr app, open your account preferences (upper left corner) and
then, under the "Panda" section, touch the id of your specific panda
(a62e6...)  This will show you a page with the wifi password for that
device.

(At first I also looked everywhere for my wireless password before I found
it there)

busyrock, 2017-12-01:
That is so covert...HaHa~

kq01526, 2017-12-10:
So you need to use the chffr app to see the WPA2 passphrase?

Is it possible to change the passphrase?

Does the Panda maybe provide a web interface to change such settings?

derhaegr, 2019-11-21:
chffr app is not supported anymore now, whats the alternative?

robbederks, 2020-01-28:
You should be able to get the wifi password from the `.get_secret()` function in the python API

closed 2020-01-28 by robbederks

UserName-wang, 2020-03-16:
Can you help me in this situation?
I cannot connect panda to my PC via USB (Win10 OS), and now I also cannot connect to panda via WIFI, because I cannot find the wifi password.

I can connect to panda before, because it's not secured.
I was trying to update my firmware, in order to connect to WIN10 OS.
And I could open the page http://192.168.0.10/ via Chrome, but after clicked the secure link in the page, then i cannot connect to panda via wifi, because i don't know the password.

Thanks!

norton-chris, 2020-05-24:
I have the same issue as you @UserName-wang, I connecting under the pair network and then when I secured it, I lost the network connection and then a new network that had a password popped up, but I wasn't given a password. Did you figure this out?

UserName-wang, 2020-05-25:
> I have the same issue as you @UserName-wang, I connecting under the pair network and then when I secured it, I lost the network connection and then a new network that had a password popped up, but I wasn't given a password. Did you figure this out?

Hope you already fixed this issue. after you unplug and re-plug your panda, you may find the network without password. and use the python script:  get_panda_password.py  , which you can download from: https://github.com/commaai/panda/tree/master/examples.

but i still have a problem with the CAN communication. I try to read a can message but I only can read the first can message, and this CAN message continusly and repeated until CAN error message came up.