commaai/panda #76: Bluetooth?

kq01526, 2017-12-10:
Hi,

why does the Panda not feature Bluetooth but only Wi-Fi?

When connecting a phone to the Panda via Wi-Fi, the phone will be unable to connect to other access points simultaneously?

With Bluetooth a phone could be connected to the Panda via Bluetooth and connect to another Wi-Fi access point simultaneously.

So why did you choose to solely rely on Wi-Fi? Why not Bluetooth in addition?

Does the Wi-Fi chip on the Panda support Bluetooth?

Will new versions of the Panda support Bluetooth?

kq01526, 2017-12-10:
Or maybe the Panda could make use of [Wi-Fi Direct](https://en.wikipedia.org/wiki/Wi-Fi_Direct)?

That way a phone could connect to the Panda in an ad-hoc mode and at the same time connect to another Wi-Fi access point simultaneously?

Or does it do that already?

energee, 2017-12-10:
It doesn't have bluetooth hardware

kq01526, 2017-12-11:
And what about the other questions?

bertramt, 2017-12-11:
Bluetooth doesn't have the data speeds capable of transferring all the data from several buses without loss.

I've heard they are trying to fix to allow data and wifi, Fairly sure dual wifi would not be possible.

dzid26, 2018-08-05:
After searching through some confusing threads about whether esp8266 supports WiFi Direct, I conclude it does not. 
And probably best it can do is access point mode, which I believe it is what it does right now.

closed 2020-01-28 by robbederks