commaai/panda #77: Latency / Frequency / Scanning rate?

kq01526, 2017-12-10:
Hi,

according to some reports and advertisements on the Internet, some OBD2 scanners seem to be too slow / too laggy for realtime measurements, i.e. when you use an app that shows the RPMs or the speed or something like that, some OBD2 scanner apparently are better at showing these in realtime because their scan rate is better than others.

Is this true?

Or does it basically depend on the car instead of the OBD2 scanner?

And what about the Panda? Will it provide the best scan rate possible, considering that it is being adviced as "panda is the nicest universal car interface ever"?

energee, 2017-12-10:
Panda is pretty comparable to kvaser as far as speed is concerned.

kq01526, 2017-12-11:
> @energee wrote:
> Panda is pretty comparable to kvaser as far as speed is concerned.

Any chance you could upload a video showing panda being used to visualize a tachometer while reving the engine?

It would be interesting to see what the lag is (if any).

In the following videos (not showing panda), it looks okay:
https://youtu.be/j5zR9QFkxGM?t=2m8s
https://youtu.be/I3PmFyYYgsk?t=22s

Whereas in some other videos (not showing panda), it looks incredibly laggy, see:
https://youtu.be/0jasYJ0TEo0?t=10s
https://youtu.be/RCSVHODoCs0?t=2m25s

Maybe it's also due to the app being used?

I don't know. I've never used an ODB2 scanner. But I would want to make sure not to get a laggy one.

So it would be interesting to see how the panda performs.

ericgingras, 2017-12-11:
For me, panda using SocketCAN is delayed about 10 seconds.

closed 2020-01-28 by robbederks