commaai/panda #79: Panda not able to read messages

bilal-alsharifi, 2017-12-24:
Hi,
I am new to Panda and I am trying to read CAN messages for my Mitsubishi Lancer 2008. I followed the instructions here:
https://community.comma.ai/wiki/index.php/Panda#Troubleshooting_Panda

But Canaba doesn't show any messages as in the screenshot. Panda is blinking red only. Can anybody tell me what I am doing wrong?
[image]

quarksteilchen, 2018-01-03:
same problem here.
panda red blinking only when no other steps are taken.

i also tried: connecting with Torque to Panda. Then it flashed wildly in different colors.
But also no Data on Cabana during this mode and connection via USB.

greerviau, 2020-03-03:
Same problem, was this issue ever resolved?

adeebshihadeh, 2020-06-09:
Closing due to inactivity. If this is still a problem, feel free to reopen.

closed 2020-06-09 by adeebshihadeh