commaai/panda #112: ModuleNotFoundError: No module named 'dfu' (on python 3.6.5)

calebmadrigal, 2018-04-19:
```
[caleb@cmbpr:panda]$ python
Python 2.7.14 (default, Apr  7 2018, 21:31:36)
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from panda import Panda
>>>
[caleb@cmbpr:panda]$ python3
Python 3.6.5 (default, Apr  6 2018, 13:56:36)
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from panda import Panda
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/panda/__init__.py", line 11, in <module>
    from dfu import PandaDFU
ModuleNotFoundError: No module named 'dfu'
>>>
```

dzid26, 2018-08-05:
It works in Python 3.7 though

KrisKusano, 2018-09-15:
Same problem

```
$ python --version
Python 3.6.5 :: Anaconda, Inc.
$ python
Python 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
[GCC 7.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from panda import Panda
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/anaconda3/lib/python3.6/site-packages/panda/__init__.py", line 11, in <module>
    from dfu import PandaDFU
ModuleNotFoundError: No module named 'dfu'
>>> 
```

Does not work on python 3.7 either.

sidlauskaslukas, 2019-05-15:
any ideas how to solve it? it does not work on python 3.7

Carver182, 2019-06-25:
Guys, I had the same issue. I solved that just using Python 2.x. It's not dedicated to Python 3.x! Either, you shout reedit the whole project to Python3 syntax like "from dfu import PandaDFU" --> "from panda.dfu import PandaDFU" and so on!

aesculus, 2019-08-07:
Any updates to this other than dropping back to 2.x? I am running 3.7.4 64 bit and I also tried @Carver182 syntax and had a similar error but this time it could not find the module panda.

robbederks, 2020-01-28:
Should work now that all code is moved to python3.7

closed 2020-01-28 by robbederks

Phalelashvili, 2020-02-14:
note: pypi package isn't moved to python3.7, install it from this repository instead