commaai/panda #126: ELM327 emulation not working

adamcb, 2018-07-16:
I recently purchased a Panda to use with Torque (among other ELM compatible applications) but I've found the ELM emulation is not working for the most part.  After multiple attempts to get Torque to work I installed an application called ELM327 Identifier and a majority of the protocols are not currently supported.  My guess this is what is keeping the application from working correctly.  Is this something that is actively being developed?

Thanks,

Adam

bertramt, 2018-07-25:
I think some vehicles still have issues.  I've never been able to make Torque work with my Panda on my 2005 Ford Explorer.  I think it might have something with the auto detection of the bus speed but I'm not sure.

vanarchist, 2018-11-21:
Are you using Torque Lite or Torque Pro? On my 2008 Ford E150, Torque Lite works at least for mode 01 PIDs. It doesn't work for any mode 22 PIDs. Torque Pro doesn't work at all.

robbederks, 2020-01-28:
The ELM327 emulation is pretty bare-bones for the pandas. This is not a big priority to fix for us, but we are happy to merge in good PRs.

closed 2020-01-28 by robbederks