commaai/panda #181: Different CAN bus index from hardware guide (Python)

gypsophlia, 2019-04-11:
I used example/can_logger.py to get can log from the ODB port, but it seems that the CAN bus index is in different order from the [Hardware Guide](https://github.com/commaai/panda/blob/master/docs/guide.pdf) which is confusing. 

From what I observed: 
Bus0 from the log means CAN-2 in Hardware Guide
Bus1 from the log means CAN-1 in Hardware Guide
Bus2 from the log means CAN-3 in Hardware Guide

Does anyone have the same problem?

vanillagorillaa, 2019-04-14:
The hardware guide starts counting bus numbers from 1, while all the code starts from 0. The hardware guide is labeling the CAN buses as if looking at the bottom of the panda. I think you are looking at it from the top and thus flipping 0 and 1

vanillagorillaa, 2019-04-24:
Closing this issue, feel free to re open if you think its wrong

closed 2019-04-24 by vanillagorillaa