commaai/panda #191: pandacan: Connection refused when connecting over WiFi

sidlauskaslukas, 2019-05-15:
Hi,

I just got my Panda and tried to use it in my car. I was able to connect to it using your app (chffr) and see the battery voltage but nothing else, then I tried to connect to it using pandacan over WiFi, but I was unable to make a connection - I am always getting an error saying "[Errno 61] Connection refused".

```
Python 2.7.16 (default, Apr 12 2019, 15:32:52)
[GCC 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.11.45.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from panda import Panda
>>> panda = Panda("WIFI")
192.168.0.10
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/panda/__init__.py", line 128, in __init__
    self.connect(claim)
  File "/usr/local/lib/python2.7/site-packages/panda/__init__.py", line 139, in connect
    self._handle = WifiHandle()
  File "/usr/local/lib/python2.7/site-packages/panda/__init__.py", line 75, in __init__
    self.sock = socket.create_connection((ip, port))
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/socket.py", line 575, in create_connection
    raise err
socket.error: [Errno 61] Connection refused
>>>
```

```
PING 192.168.0.10 (192.168.0.10): 56 data bytes
PING 192.168.0.10 (192.168.0.10): 56 data bytes
64 bytes from 192.168.0.10: icmp_seq=0 ttl=128 time=2.119 ms
64 bytes from 192.168.0.10: icmp_seq=1 ttl=128 time=12.972 ms
64 bytes from 192.168.0.10: icmp_seq=2 ttl=128 time=1.175 ms
64 bytes from 192.168.0.10: icmp_seq=3 ttl=128 time=1.330 ms
64 bytes from 192.168.0.10: icmp_seq=4 ttl=128 time=12.646 ms
64 bytes from 192.168.0.10: icmp_seq=5 ttl=128 time=9.967 ms
^C
--- 192.168.0.10 ping statistics ---
6 packets transmitted, 6 packets received, 0.0% packet loss
```

What I'm doing wrong here?

li895, 2019-11-13:
I ran into the same issue. It turns out you need to first enable "secure mode" for the WiFi connection to work in pythonk. When you've connected to the WiFi network, navigate to http://192.168.0.10 and turn on "secure mode" link (before you do this make sure you've written down the password for the device first... I think the way you find the password is using one of the example scripts from the library). Once you turn this on your WiFi will drop, and you will have to reconnect using WPA/WPA2 password authentication. Once that is connected, if you go back to the website it should say "in secure mode" and your python script should work.

sidlauskaslukas, 2019-11-13:
@li895 in my case it was the Apple's USB-C adapter. You have always connect USB-C adapter first and then Panda. You can not connect them all together at the same time.

closed 2020-01-28 by robbederks

UserName-wang, 2020-03-16:
@li895 , hi, I enabled the "secure mode", before I set the password. how can I find the default password now? and i cannot connect panda to WIN10 OS. thanks!

li895, 2020-04-15:
Once you reboot the Panda, the secure mode should go away and you can start over.