commaai/panda #273: windows Python 2.7, having issue when running can_logger.py

zhouhao0605, 2019-09-19:
I'm running the same code, it works in Ubuntu, but fails in my laptop, which is Windows 10. I cannot understand why. I had every packages installed though.

D:\Projects\panda\venv\Scripts\python.exe D:/Projects/panda/python/can_logger.py
Traceback (most recent call last):
  File "D:/Projects/panda/python/can_logger.py", line 6, in <module>
    from panda import Panda
  File "D:\Projects\panda\venv\lib\site-packages\panda\__init__.py", line 7, in <module>
    import usb1
  File "D:\Projects\panda\venv\lib\site-packages\usb1\__init__.py", line 61, in <module>
    from . import libusb1
  File "D:\Projects\panda\venv\lib\site-packages\usb1\libusb1.py", line 199, in <module>
    libusb = _loadLibrary()
  File "D:\Projects\panda\venv\lib\site-packages\usb1\libusb1.py", line 173, in _loadLibrary
    return dll_loader('libusb-1.0' + suffix, **loader_kw)
  File "C:\Python27\Lib\ctypes\__init__.py", line 366, in __init__
    self._handle = _dlopen(self._name, mode)
WindowsError: [Error 126] The specified module could not be found

zhouhao0605, 2019-09-19:
I did some research online, looks like it is related to C++ compiler? I'm not sure. Can anyone provide some insights?

robbederks, 2020-01-28:
It cannot seem to find libusb on your windows pc. Make sure that this is installed correctly and try again.

closed 2020-01-28 by robbederks

dzid26, 2020-05-01:
I don't think there is a concept of installing [libusb-1.0.dll](https://github.com/libusb/libusb/releases) on Windows. 
> python-libusb1 can be expected to work on:
> Windows [1] native dll or via Cygwin

- I tried "installing" it with Cygwin but it didn't install anything it. The dll was not downloaded anywhere.
- `pip install libusb1`  [also doesn't provide it](https://github.com/vpelletier/python-libusb1/issues/33) 
- [Putting libusb-1.0.dll next to the script](https://github.com/vpelletier/python-libusb1/issues/7) worked though

Can libusb-1.0.dll be provided in **panda** repo?