commaai/panda #304: CAN communication not working when compiled from master

xtensa, 2019-10-26:
After cloning master branch and compiling building firmware for panda (under board directory), Panda is successfully flashed. However, green light is flickering quickly. WiFi and USB communication works fine, but CAN communication isn't working.

robbederks, 2019-10-27:
If the green LED is flashing quickly, it looks like the flashing wasn't successful, and the panda is stuck in the bootloader mode.
Try flashing the panda again, with the command `make`, and if that doesn't fix is, `make recover`.

In normal operation, the LED should pulse red.

xtensa, 2019-10-27:
I don't think it's stuck in bootloader mode, because as I said: WiFi is working, I can communicate with USB and make API calls. So it is not in bootloader mode. I tried reflashing with 'make' or manually multiple times, but the result is the same. I also can revert to any stock FW without any problem - this is not the issue.

robbederks, 2019-10-28:
Well, the Wifi is processed on the separate ESP module, and the bootloader also implements some USB calls (see `board/spi_flasher.h`). This is also the only place in the code that blinks the green LED quickly (https://github.com/commaai/panda/blob/de18a7ef1a01980faee44bdb08b29bb0ebf381e5/board/spi_flasher.h#L322).

What do you mean that you can revert to any stock FW? Are you using modified code?

xtensa, 2019-10-29:
> What do you mean that you can revert to any stock FW? Are you using modified code?

I meant that I can load any "official" downloaded firmware using this command:
`python3 -c "from panda import Panda; Panda().flash('panda.bin')"`
and all those firmwares worked.

Finally after several 'make recover' newly compiled firmware worked. Now, if I recompile FW even with my mods, it still loads properly with just "make". Cannot explain this inconsequence, but important think is that the problem is solved.

closed 2019-10-29 by xtensa