commaai/panda #305: `make` from master (for board) does not work

NamoDev, 2019-10-28:
Hey there! This might be a simple question (or perhaps a dependency issue), but I'm trying to build the firmware as is from this repo. Cloned it down, followed the README, and it seems that there are errors within `esptool.py`?

Any pointers on how to go from here? Thanks!

```
PYTHONPATH=../ python -c "from python import Panda; Panda().flash('obj/panda.bin')"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/secondary/DevProjects/panda/python/__init__.py", line 13, in <module>
    from .esptool import ESPROM, CesantaFlasher  # noqa: F401
  File "/secondary/DevProjects/panda/python/esptool.py", line 842
    print('Downloading %d bytes at %08x...' % (size, offset), end=' ')
                                                                 ^
SyntaxError: invalid syntax
build.mk:36: recipe for target 'flash' failed
make: *** [flash] Error 1
```

rbiasini, 2019-10-28:
Are you using Python 3.7 by default? You need to.

NamoDev, 2019-10-28:
Ah, that seems to be the problem. The documentation doesn't seem to point this out... would you mind if I send in a pull request later for documentation updates?

closed 2019-10-29 by NamoDev