commaai/panda #310: Received message lag

Prj1508, 2019-10-29:
I have panda and giraffe combination to connect into Chrysler front camera module. 

https://comma.ai/shop/products/panda-obd-ii-dongle
https://comma.ai/shop/products/giraffe

I have ubuntu 16.04 machine and I am connected to Panda thru USB. I am using socketcan running at 500 baud. I am currently experiencing delays/lag in receiving the CAN messages. I am testing onboard a vehicle and when I turn the steering wheel for example, there is 5-10 seconds delay in updating the data field of the respective message in socketcan(in candump or kayak). I tested with Vector drivers and the update is almost instantaneous. I am wondering what could be causing these delays in panda and how can I resolve this issue. Are there any additional settings I will have to do? Is 120ohm terminal resistance already part of Giraffe/Panda? Appreciate any help

thank you

AnEthernetCable, 2020-08-24:
This still appears to be an issue on the latest hardware and software. Any updates on this?

maoweiming, 2021-04-27:
You can1t send messages on CAN?You can tell me how to do it

closed 2023-06-17 by adeebshihadeh