commaai/panda #316: Can1t send messages on CAN

Lviktorix, 2019-11-11:
I am trying to send out a message from the panda on CAN bus, in my program I connect to panda via USB, I am seting Safety_Alloutput mod, i am setting the can bus speed and the loopback mod on false, but when i am trying to send a message with the can_send() there is nothing on the bus. 
I tried it with loopback mod on too, but still got nothing. 
I am sure it is not a hardware proglem, i can send messages on the bus with other CAN tools. 
I tried all of the 3 busses.
I can receive messages without any problem this is why i think it is weird.
Can anyone help me?

tareeqav, 2020-05-02:
@Lviktorix did you ever solve this? I am having the same issue

maoweiming, 2021-04-27:
did you ever solve this? I am having the same issue

maoweiming, 2021-04-27:
did you ever solve this? I am having the same issue.

flapse, 2021-11-05:
I believe this got to do with the safety settings. In the release firmware of panda you can only send limited instructions to the outputs:

quote (from README.md):
"
```
Safety Model

When a panda powers up, by default it's in SAFETY_SILENT mode. While in SAFETY_SILENT mode, the buses are also forced to be silent. In order to send messages, you have to select a safety mode. Currently, setting safety modes is only supported over USB. Some of safety modes (for example SAFETY_ALLOUTPUT) are disabled in release firmwares. In order to use them, compile and flash your own build.

Safety modes optionally supports controls_allowed, which allows or blocks a subset of messages based on a customizable state in the board.
```
"

Not sure what the subset of messages is allowed with release software...

I am having the same problems with my white panda. I am trying to run the "query_vin_and_stats.py" program of George (examples/query_vin_and_stats.py) but there is no response to any query.

I have tried to compile my own firmware (with #define DEBUG in board/config.h) but this still does not give me access to the outputs. If anyone has an idea what compiler settings to do to enable the outputs, please post your solution below. (BTW I have written down my struggles here: https://github.com/commaai/panda-artifacts/issues/1)

closed 2023-08-06 by adeebshihadeh