commaai/panda #372: ISO26262: safety modes should white-list allowed messages instead of blacklisting the disallowed ones

rbiasini, 2019-11-13:


jyoung8607, 2019-11-13:
A white-list seems a laudable goal, but I'm ass-u-ming you mean to apply it on EON's TX/RX only. A white-list on forwarded traffic seems like it would be a serious maintenance headache.

Per our DM discussion a couple weeks ago, the forwarding hooks are currently entangled with the EON RX/TX hooks in a very counter-intuitive fashion, and I think that would have to be fixed first.

rbiasini, 2019-11-13:
Correct! Tx hook should not apply to fwd'ed messages. Or at least not like it is applied now (maybe we can check of the rx'd msg and fw'd message are any different in content. In case, then we apply the tx hook.

rbiasini, 2019-11-16:
@jyoung8607 , per our discussion ^^

- skip tx hook for fwd'd msgs: https://github.com/commaai/panda/commit/4d1edc068b507fe3fb4bf7a309bb9a45af547a7d
- whitelist msgs: https://github.com/commaai/panda/pull/381/files

closed 2019-11-17 by rbiasini