commaai/panda #386: Set Safety Mode 0x1337 crashes white panda

geohot, 2019-11-19:
On master, panda fw: v1.6.3-EON-8a044b34-DEBUG

root@localhost:~/one/panda/board$ python
Python 3.7.3 (default, Apr 24 2019, 11:21:56) 
[Clang 8.0.2 (https://android.googlesource.com/toolchain/clang 40173bab62ec7462 on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import
KeyboardInterrupt
>>> import panda
>>> panda.Panda().set_safety_mode(0x1337)
opening device 540031000651363038363036 0xddcc
connected

Panda is hung, LED is no longer blinking.

jyoung8607, 2019-11-19:
That's intentional, and should have put something in the debug console buffer. See this commit:

https://github.com/commaai/panda/commit/0e72c18378b04d22ea502027e9d5cc0d0e3a25fe

It was part of some cleanup of the init and safety setting code. It carried forward the old init-time behavior of stopping if the safety mode could not be set, and made it applicable for all failures not just init-time.

https://github.com/commaai/panda/blob/ff86db65538a2ac8bace47a1790fb7d8fd933287/board/main.c#L772

If you don't want to halt/spin on setting an invalid safety mode, then don't. But, it should probably be modified to set SAFETY_NOOUTPUT instead.

jyoung8607, 2019-11-20:
Following up to myself, because I knew 0x1337 looked familiar.... the Panda safety list was aligned with cereal a while back, and SAFETY_ALLOUTPUT was renumbered from 0x1337 to 17. I didn't realize you intended to set a valid working safety mode when I replied earlier, I thought you were fuzzing, but I think the substance of my reply applies anyway.

geohot, 2019-11-20:
Crashing is never okay. No output is fine.

rbiasini, 2019-11-20:
yeah, I can make it default to noOutput is safety mode setting fails and then ok to hang if that one fails too.

rbiasini, 2019-11-20:
https://github.com/commaai/panda/commit/a54b86c4baa8ee935c659436abde6ba94715c439

closed 2019-11-20 by rbiasini