commaai/panda #400: panda schematic

dlednik, 2019-12-11:
Is it possible to get the schematics for panda OBD-II?

robbederks, 2019-12-17:
All hardware designs are closed-source, although you can find the pinout for the OBD-II connector here: https://medium.com/@comma_ai/a-panda-and-a-cabana-how-to-get-started-car-hacking-with-comma-ai-b5e46fae8646

closed 2019-12-17 by robbederks

dzid26, 2019-12-22:
Hopefully reverse engineering is not frowned upon
[image]
[image]
[image]
[image]
[image]
[image]
[image]

dlednik, 2019-12-23:
@dzid26 was thinking in that direction but decided it might be better to make a new one based on ST32 that has arduino bootloader available :)

dzid26, 2019-12-23:
Yeah, it depends on what you need it for. 

My story was that I just needed a single digital output, but I found no direct connections between the board and OBD pins. Then I figured GMLAN or LIN could work as digital out to some degree.  Later, luckily, I noticed Comma already implemented the digital outs using those pins.

martiinezz, 2020-02-11:
Did you guys recognize all parts? I know can transceiver is NXP and LIN too. Looks rather simple. I was thinking about doing some proper diagnostics interface for cars but this form factor is too small IMHO.

dlednik, 2020-02-11:
If you gonna build your own part# don't matter that much as you can use whatever you want :)

I did not had time yet to work on this. But if there is interest for an OS project we can collaborate. Need to finish a webstore for client first but after that I can dedicate more time for this.

martiinezz, 2020-02-12:
I know - to be hones I don't even like that Panda is so small. Already recognized that most elements are build (including layout) to chips manufacturers recommendations. Should be quite easy to put it all together. I also noticed that there has been some changes compared to what is shown in leaflet that comes with Panda. For example current sensor and LDO for ARM is now on the other side than ESP. Also GM LAN moved so ST buck converter can easier dissipate heat thru vias.

luannbrandao, 2023-10-01:
> @dzid26 was thinking in that direction but decided it might be better to make a new one based on ST32 that has arduino bootloader available :)

you did it?

dlednik, 2023-10-14:
> > @dzid26 was thinking in that direction but decided it might be better to make a new one based on ST32 that has arduino bootloader available :)
> 
> you did it?

no, not enough interest and still busy with other stuff, but like I said open to start and collaborate on OS variant to tap into cars OBD2 data

martiinezz, 2023-10-20:
I have identified all components here: https://github.com/martiinezz/panda/tree/master/white-hw - most other components are exactly the same as listed chip makers reference designs. Easy to reproduce. Winter is coming and I finally might start KiCad design

dzid26, 2023-10-20:
>  Easy to reproduce. Winter is coming and I finally might start KiCad design

https://github.com/lukasloetkolben/OpenpilotHardware/tree/main/BlackPanda/kicad_project
Maybe black&white panda breed. ;)

martiinezz, 2023-10-20:
I don't have black panda at hand so unable to reproduce :(