commaai/panda #416: safety_tesla.h shows issues in debug_console.py

dzid26, 2019-12-30:
**Describe the bug**
When Tesla safety is "hooked", the printing in debug_console doesn't work and doesn't recover until reset.

**How to reproduce or log data**
I am developing new car safety based in safety_tesla.h, so it is triggered according to my code, but it should also work by replacing, let's say {SAFETY_YOURVEHICLE &yourvehicle}, with   {SAFETY_YOURVEHICLE, &tesla_hooks} in safety.h.
Add some printing to _tesla_init_ E.g. puts("Initialization\n");.  
Flash panda.
Run ./debug_console.py
And then raise ignition_line high to enable OpenPilot.

**Expected behavior**
I am expecting to see the following printing messages after launching openpilot:
```
EON hasn't sent a heartbeat for 0x0000000f seconds. Safety is set to SILENT mode.
disable power savings
Invalid ESP/GPS mode
Initialization
EON hasn't sent a heartbeat for 0x00000005 seconds. Safety is set to SILENT mode.
```

**Observed behavior**
Instead, I see:
```
EON hasn't sent a heartbeat for 0x0000000f seconds. Safety is set to SILENT mode.
disable power savings
Invalid ESP/GPS mode
```
And the no matter what happens next, the printing in the console is not available, unless restarting panda.

./health_test.py seems to be unaffected by the problem, so I don't think panda is in reset.

This whole issue is probably a symptom of some other problem so I will be trying to reduce tesla_safety.h in order to debug it

dzid26, 2019-12-30:
gmlan_switch_init(1); is causing the problem. Probably some conflict with interrupts or timers. 
Unfortunately for me, the reason I am using tesla_safety _was_ the gmlan_switch functionality.  So now I need to figure out why gmlan switch doesn't work and why does it break debug_console printing

dzid26, 2019-12-30:
I was able to fix GMLAN switch and broken debug_console.py symptom by adding `REGISTER_INTERRUPT(TIM4_IRQn, TIM4_IRQ_Handler, 40000U, FAULT_INTERRUPT_RATE_GMLAN)  
`  
in
`gmlan_switch_init(int timeout_enable) `

I don't exactly know what are the intentions of call rate and fault rate parameters, so I just copied it from bitbang_gmlan.

rbiasini, 2019-12-30:
@robbederks , can you please have a look?

robbederks, 2020-01-10:
Just reading this now, thanks for debugging it so far already! Will look into it myself later this week, but you're probably right about not registering the interrupt. Should not cause the debug console to go dark though, have to look into that.

closed 2020-01-28 by robbederks