commaai/panda #420: panda use Socket CAN can't Send Data

boyuchen1029, 2020-01-07:
Host PC environment: Ubuntu 16.04
I use Socket CAN through USB and CAN0 can receive CAN bus data,
But can't use Socket CAN send CAN bus data, and also use Python can't send any CAN bus data.

ozzdemir, 2020-04-20:
Any update in this issue @boyuchen1029 ?

ozzdemir, 2020-04-21:
I found some hint I guess about the issue. When 500 kbps is used as bitrate, panda cannot communicate. I tested it with another can adapter and also within panda also. Used following code to test within panda (Using CAN0 and CAN1) @robbederks even when safety is set to alloutput.

```
#!/usr/bin/env python3
import sys
from panda import Panda
import time
try:
  print("Trying to connect to Panda over USB...")
  p = Panda()
except AssertionError:
  print("USB connection failed.")
p.set_can_speed_kbps(0, 125)
p.set_can_speed_kbps(1, 125)
p.set_safety_mode(Panda.SAFETY_ALLOUTPUT) # Activating output
while True: # Periodically sending and receiving.
    p.can_send(0x248, b"\x11\x22\x33\x44\xFF\x00\x00\xFF", 1)
    print(p.can_recv())
    time.sleep(1) 
```
When 125 kbps is used as can speed, can_recv() return the message sent, but when 500 kpbs is used, an empty list is returned. Connection between busses made using jumpers as in the figure below. Note that same result is observed when another can adapter is used. Is there such a limitation of black panda/harness? I couldn't find such a document.
[image]

robbederks, 2020-04-24:
Do you have any termination resistors on the CAN bus? The speed of the bus should not influence the workings at all, but it might just barely work with very high resistance between the H and L wires.
Try measuring the resistance between the pairs, should be 60 ohms

ozzdemir, 2020-04-29:
That was the case, solved by terminating bus.

tareeqav, 2020-05-02:
@robbederks @ozzdemir I am trying to do the same thing, only I do not have the standard panda connector, I am working with this [one](https://www.amazon.com/iKKEGOL-Connector-Diagnostic-Extension-Pigtail/dp/B07F16HC12)

I've plugged this into the panda, connected lines 11 and 13 together with a 120 ohm resistor in between, and 3 and 6 together (with the same 120 ohm resistor in between).

Based on [this](https://camo.githubusercontent.com/b442f9e8ee29fd2330a2fd906ccd8027463ff23e/68747470733a2f2f692e696d6775722e636f6d2f7736746e6c44712e706e67), 3 and 11 are CAN2 H and L respectively while 6 and 14 are CAN1 H and L.

Using a multimeter I measured 6.2 K ohms, but when I run the script from @ozzdemir I am still seeing empty lists being printed.

Any thoughts?

robbederks, 2021-05-17:
6.2k is too high for CAN to function. There should be 60 ohms termination between the bus lines.
Closing this issue now, seems like it's solved.

closed 2021-05-17 by robbederks