commaai/panda #430: Can't build on Mac

bthaler1, 2020-01-26:
I'm trying to build from source on Mac. I get the following error running "make" from /board:

arm-none-eabi-gcc -g -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -I inc -I ../ -nostdlib -fno-builtin -std=gnu11 -Os -Tstm32_flash.ld "-DALLOW_DEBUG" -o obj/startup_stm32f413xx.o -c startup_stm32f413xx.s
echo "const uint8_t gitversion[] = \"v1.7.3-DEV-f67ec28e-"DEBUG"\";" > obj/gitversion.h
../crypto/getcertheader.py ../certs/debug.pub ../certs/release.pub > obj/cert.h
arm-none-eabi-gcc -MT obj/main.panda.o -MMD -MP -MF generated_dependencies/main.Td -g -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -I inc -I ../ -nostdlib -fno-builtin -std=gnu11 -Os -Tstm32_flash.ld "-DALLOW_DEBUG" -o obj/main.panda.o -c main.c
arm-none-eabi-gcc -Wl,--section-start,.isr_vector=0x8004000 -g -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -I inc -I ../ -nostdlib -fno-builtin -std=gnu11 -Os -Tstm32_flash.ld "-DALLOW_DEBUG" -o obj/panda.elf obj/startup_stm32f413xx.o obj/main.panda.o
arm-none-eabi-objcopy -v -O binary obj/panda.elf obj/code.bin
copy from `obj/panda.elf' [elf32-littlearm] to `obj/code.bin' [binary]
SETLEN=1 ../crypto/sign.py obj/code.bin obj/panda.bin ../certs/debug
signing 33504 bytes
hash: 0b9fbaafdca9fb8241e24b58aa6e7a99492be036
du: illegal option -- b
usage: du [-H | -L | -P] [-a | -s | -d depth] [-c] [-h | -k | -m | -g] [-x] [-I mask] [file ...]
/bin/sh: line 0: [: -ge: unary operator expected
PYTHONPATH=../ python3 -c "from python import Panda; Panda().flash('obj/panda.bin')"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/bthaler/Desktop/Comma_ai/panda/python/__init__.py", line 154, in __init__
    self.connect(claim)
  File "/Users/bthaler/Desktop/Comma_ai/panda/python/__init__.py", line 199, in connect
    assert(self._handle != None)
AssertionError
make: *** [flash] Error 1

robbederks, 2020-01-26:
Hi,

This error happens in the flashing portion of the script, which tries to find a connected panda to flash.
If you're just trying to compile the code and not flash the panda, this can safely be ignored.
If you are however trying to flash a panda, make sure that it shows up as a USB device when connected (run `lsusb` and check for a device with ID `bbaa:ddcc`). If it is, you probably need more privileges to access it (`sudo make` or udev rules? Not sure on mac.)

bthaler1, 2020-01-26:
Thanks for the help. I tried the udev rules, no difference. White Panda doesn't show up as a USB device. It must be bricked. I tried with my Gray Panda and was able to run sudo make, then the Panda rebooted and never reconnected. Now it also refuses to show up as a USB device. Something in the firmware has bricked both of my Pandas. White Panda briefly has solid blue LED which then flashes a few times and then nothing. Gray Panda shows no LED.

robbederks, 2020-01-27:
I see. Can you try 'sudo make recover' and see if that does anything? Do you have a panda paw?

bthaler1, 2020-01-27:
Make recover gives a similar error, since the Panda can't be found as a USB device:

PYTHONPATH=../ python3 -c "from python import Panda; Panda().reset(enter_bootloader=True)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/bthaler/Desktop/Comma_ai/panda/python/__init__.py", line 154, in __init__
    self.connect(claim)
  File "/Users/bthaler/Desktop/Comma_ai/panda/python/__init__.py", line 199, in connect
    assert(self._handle != None)
AssertionError
make: [recover] Error 1 (ignored)
sleep 1.0
"dfu-util" -d 0483:df11 -a 0 -s 0x08004000 -D obj/panda.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
make: *** [recover] Error 74

I do not have a Panda Paw. The Panda's Wifi is actually still working, but it's been secured and I have no way to get the password since the Chffr app is no longer supported.

bthaler1, 2020-01-27:
Update: I managed to connect to the Panda via WiFi and upload panda.bin and user1.bin and user2.bin via the WiFi flashing instructions here: https://community.comma.ai/wiki/index.php/Panda_Flashing, but the Panda still will not connect via USB. I'm not sure if it's related but on the Panda's web page, it shows:
st version:     
esp version:    v1.7.0-DEV-c108df4a-RELEASE
esp flash file: user2.bin

I tried several different firmware versions, and all had the same result.

robbederks, 2020-01-27:
Did you modify the code before flashing the pandas? Normally they should be very resilient against bricking, I don't see immediately how a successful flashing can cause this without modifying the code.
Anyway, if they do not come up as USB devices anymore in any scenario (different pc, re-powering them, other cable, ...), you can always order a panda paw to put them into DFU mode and unbrick them with `sudo make recover`.

If they actually got bricked without modifying code, I'd be very interested in a detailed step list that actually bricked the pandas for you to see if I can replicate this. This should not happen

robbederks, 2020-01-27:
The reason the wifi still works is that that code is run on a separate processor, and is not dependent on the STM code to boot.

bthaler1, 2020-01-27:
I did not modify the code at all before flashing. I just cloned the repo, ran get_sdk_mac.sh (which doesn't work btw), and then tried running various combinations of make and make recover, make clean etc. I have tried 2 different macs and 2 different cables.

I also don't understand how the flash via WiFi can be successful and still not work. Is the "st version" on the Panda's web page supposed to be blank?

robbederks, 2020-01-28:
I find it concerning that these procedures bricked your panda. Can you try on a linux (ubuntu, ...) machine and see if anything comes up with `lsusb`?

bthaler1, 2020-01-29:
lsusb shows nothing on Linux as well.  Strangely, the Mac version of lsusb shows this:
2020-01-28 20:13:58.297 system_profiler[1102:135524] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be

bthaler1, 2020-01-29:
So I managed to get the White Panda into DFU mode by pulling boot0 high according to this document: https://github.com/commaai/panda/blob/master/docs/guide.pdf

I managed to get make recover to flash the Panda and it indicated the flash was successful, but upon reboot there was no difference. No USB device, still can't run make. I managed to get it into DFU mode a few times with the same result and now it won't even go into DFU mode at all.

The Grey Panda refuses to go into DFU mode at all via this method so it appears to be completely dead.

robbederks, 2020-02-05:
Very strange. Have you tried flashing it on a linux machine? Maybe it is something mac related.

gregjhogan, 2020-05-17:
This issue also applies to ubuntu 20.04

robbederks, 2020-06-02:
Should be fixed in #549

closed 2020-06-02 by robbederks