commaai/panda #449: Some panda's don't go into dsu mode when trying to run openpilot/devel

pd0wm, 2020-02-24:
We've gotten multiple reports of pandad getting stuck in "waiting for DSU" when trying to reflash the bootstub. This seems very repeatable for those people. Haven't reproduced yet.

jyoung8607, 2020-04-16:
Per @robbederks / f07a6ee we need to bounce through the bootstub first to reach DFU mode from release builds of Panda firmware. I haven't examined the boardd startup process to see why this isn't happening successfully during normal startup (or if this is something that should work by design) but making the same fix to `make recover`, which uses different code in `build.mk` to enter DFU, I was eventually able to get an affected user fixed.

This did take two tries to get into DFU, or there may be another state transition required here that I don't see or understand that was necessary to reach DFU mode.

```
root@localhost:/data/openpilot/panda/board$ make recover
arm-none-eabi-gcc -MT obj/bootstub.panda.o -MMD -MP -MF generated_dependencies/bootstub.Td -g -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -I inc -I ../ -nostdlib -fno-builtin -std=gnu11 -Os -Tstm32_flash.ld "-DEON" "-DALLOW_DEBUG" -o obj/bootstub.panda.o -c bootstub.c
arm-none-eabi-gcc -g -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -I inc -I ../ -nostdlib -fno-builtin -std=gnu11 -Os -Tstm32_flash.ld "-DEON" "-DALLOW_DEBUG" -o obj/sha.panda.o -c ../crypto/sha.c
arm-none-eabi-gcc -g -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -I inc -I ../ -nostdlib -fno-builtin -std=gnu11 -Os -Tstm32_flash.ld "-DEON" "-DALLOW_DEBUG" -o obj/rsa.panda.o -c ../crypto/rsa.c
arm-none-eabi-gcc -g -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -I inc -I ../ -nostdlib -fno-builtin -std=gnu11 -Os -Tstm32_flash.ld "-DEON" "-DALLOW_DEBUG" -o obj/bootstub.panda.elf obj/startup_stm32f413xx.o obj/bootstub.panda.o obj/sha.panda.o obj/rsa.panda.o
arm-none-eabi-objcopy -v -O binary obj/bootstub.panda.elf obj/bootstub.panda.bin
copy from `obj/bootstub.panda.elf' [elf32-littlearm] to `obj/bootstub.panda.bin' [binary]
PYTHONPATH=../ python3 -c "from python import Panda; Panda().reset(enter_bootstub=True)"
opening device 180045001651383432363133 0xddcc
connected
opening device 180045001651383432363133 0xddee
connected
PYTHONPATH=../ python3 -c "from python import Panda; Panda().reset(enter_bootloader=True)"
opening device 180045001651383432363133 0xddee
connected
sleep 1.0
"tools/dfu-util-aarch64" -d 0483:df11 -a 0 -s 0x08004000 -D obj/panda.bin
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

tools/dfu-util-aarch64: Invalid DFU suffix signature
tools/dfu-util-aarch64: A valid DFU suffix will be required in a future dfu-util release!!!
tools/dfu-util-aarch64: No DFU capable USB device available
make: *** [build.mk:48: recover] Error 74
root@localhost:/data/openpilot/panda/board$ make
PYTHONPATH=../ python3 -c "from python import Panda; Panda().flash('obj/panda.bin')"
opening device 180045001651383432363133 0xddcc
connected
flash: bootstub version is v1.7.4-DEV-f67ec28e-RELEASE
flash: unlocking
flash: erasing
flash: flashing
flash: resetting
opening device 180045001651383432363133 0xddee
connected
root@localhost:/data/openpilot/panda/board$ make
PYTHONPATH=../ python3 -c "from python import Panda; Panda().flash('obj/panda.bin')"
opening device 180045001651383432363133 0xddee
connected
flash: main version is v1.7.4-DEV-f67ec28e-RELEASE
flash: bootstub version is v1.7.4-DEV-f67ec28e-RELEASE
flash: unlocking
flash: erasing
flash: flashing
flash: resetting
opening device 180045001651383432363133 0xddee
connected
root@localhost:/data/openpilot/panda/board$ make recover
PYTHONPATH=../ python3 -c "from python import Panda; Panda().reset(enter_bootstub=True)"
opening device 180045001651383432363133 0xddee
connected
opening device 180045001651383432363133 0xddee
connected
PYTHONPATH=../ python3 -c "from python import Panda; Panda().reset(enter_bootloader=True)"
opening device 180045001651383432363133 0xddee
connected
sleep 1.0
"tools/dfu-util-aarch64" -d 0483:df11 -a 0 -s 0x08004000 -D obj/panda.bin
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
[... 23 lines cut ...]
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

tools/dfu-util-aarch64: Invalid DFU suffix signature
tools/dfu-util-aarch64: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 13508
Download        [=========================] 100%        13508 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
```

pd0wm, 2020-05-15:
This seems to be fixed now.

closed 2020-05-15 by pd0wm