commaai/panda #471: Panda has no instruction to install

stplaydog, 2020-03-14:
When I get the panda shipped to me, there is only one link: comma.ai/setup, and following this link, there is no useful information to set up panda.

robbederks, 2020-03-14:
A standalone panda is mostly a development tool, and the code is pretty self-explanatory for most use-cases. What are you planning on using the panda for?

stplaydog, 2020-03-14:
Thanks for response, I want to use panda to record my car's usage stats. My car is a Toyota corolla 2020 model. I tried with the instruction on the panda/docs folder, but it seems that I will need another panda paw to get the wifi password. Let me see if there is anything else I need for help, which I will record here.

robbederks, 2020-03-15:
Ok, I will close this issue for now.

closed 2020-03-15 by robbederks