commaai/panda #475: LIN read not working

ChristopherReeves, 2020-03-28:
kline_ll_recv method hangs. controlRead 0xe0 never returns anything. I am able to send write to it, Arduino has no problem picking up Panda's messages.
two pandas connected to each other can't read each other's messages either.
tried on both LIN buses.

Phalelashvili, 2020-04-01:
after some painful debugging, we figured out that `uart_interrupt_handler` is not fired for USART3 and UART5. if I call `dma_rx_init` before trying to read, I can read but this approach is impractical

gregjhogan, 2020-07-23:
I am not sure what your issue is, but for me it definitely works hooking two pandas together.

I know the loopback test works, so that is a known working example:
https://github.com/commaai/panda/blob/master/tests/loopback_test.py

Phalelashvili, 2020-07-23:
> I am not sure what your issue is, but for me it definitely works hooking two pandas together.
> 
> I know the loopback test works, so that is a known working example:
> https://github.com/commaai/panda/blob/master/tests/loopback_test.py

issue was created before your patch

gregjhogan, 2020-07-23:
> issue was created before your patch

what patch are you referring to?

Phalelashvili, 2020-07-23:
> > issue was created before your patch
> 
> what patch are you referring to?

https://github.com/commaai/panda/commit/31f8a0d862f699d0d06593085aa16a8cd657bccc
Interrupts were not enabled before

gregjhogan, 2020-07-23:
good call, this was most likely fixed the issue:

https://github.com/commaai/panda/commit/31f8a0d862f699d0d06593085aa16a8cd657bccc#diff-9cd7fc03cea8ce3991e180de7c34ed2bR294-R296

thanks!

closed 2020-07-23 by gregjhogan