commaai/panda #497: Toyota Interceptor - Threshold Limit

iHelp101, 2020-04-12:
**Issue:** I have been having issues with my pedal for a while (I bought another to make sure it wasn't faulty). It will engage for 10-15 seconds and then steering would be temporarily unavailable until faulting. With the pedal's Ethernet cables unplugged no issues occurrs.

**Conclusion:** After testing it appears this is caused by the interceptor going over TOYOTA_GAS_INTERCEPTOR_THRSLD's 475 value in my 2018 Toyota Corolla LE, which results in controls being disabled. Below is an output from a 5 minutes drive that showcases the threshold when over 475 (Max is 491).

[image]

**Possible Solution:** Increase TOYOTA_GAS_INTERCEPTOR_THRSLD to 500. Due to this being Panda safety code I wanted to mention this and see others comments/concerns before making a pull request. Thank you.

robbederks, 2020-04-12:
I do not believe this threshold to be the issue, since it is not empirically chosen and should match the logic in openpilot. The safety code checks if the pedal is pressed, in which case controls are not allowed anymore and the system should disengage.
Openpilot should also disengage at that same point though, which I am suspecting it does not?
Are you running a fork or do you have local changes?

iHelp101, 2020-04-13:
Yes, OP disengages and causes a steering fault (Only when pedal Ethernet is plugged in). I am using stock OP. If I change the value to 500 OP is happy.

iHelp101, 2020-04-13:
Here is an example Cabana. The fault and forced disengagement happens at 94 seconds. Right before the fault the pedal for some reason has a rise that is triggering this rising edge code for the interceptor. This happens every time about 10-15 seconds into OP being engaged with the pedal on my Corolla. This occurs on two pedals (One I sourced myself and another I purchased from a community member). 

[image]

https://my.comma.ai/cabana/?route=b2539da15e902d80%7C2020-04-11--00-00-53&exp=1618278368&sig=NnQwxHXD2%2FpsnIXJHQmgnrQIi1Esoe%2BxOMqNQODns0Q%3D&max=5&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Fb2539da15e902d80%2Fd0352c4789a50758c0a4a784b9127900_2020-04-11--00-00-53

robbederks, 2020-04-13:
Thanks for the detailed write-up. I'll look into this more later to see how the two limits relate. Ideally, openpilot should disengage at the same point or slightly quicker than the panda safety

iHelp101, 2020-04-14:
Thank you. For now I just am using stock OP with the one Panda change I mentioned (Threshold is set to 500). Hopefully a fix is up streamed soon.

robbederks, 2020-04-15:
Possible fix in PR #507

robbederks, 2020-04-16:
Fix is merged in to the master branch, seems to work in our car.
Closing the issue for now, let me know if it still doesn't work.

closed 2020-04-16 by robbederks