commaai/panda #512: Failed to connect ESP8266 when using flash_release()

ozzdemir, 2020-04-20:
I was testing black panda with another can adapter. Panda was able to receive the message but failing to send ack I guess since other adapter was keep sending non-stop. When I tried to send from panda, none was observed from other can adapter. Thats why I tried to update firmware using
`$ PYTHONPATH="" sudo python3.7 -c "import panda; panda.flash_release()"`
But gotten following output, do you have any idea what is the reason for this error?
Also any idea about not being able to send any can message? 
Note that I was having error similar to #51 and solved it by make recover if it matters.

Thanks in advance.
```
Using panda with serial 41004c000251373034353037
Fetching latest firmware from github.com/commaai/panda-artifacts
Fetching firmware from https://github.com/commaai/panda-artifacts/raw/master/panda-v1.7.3-DEV-d034f3e9-RELEASE.zip
File Name                                             Modified             Size
bootstub.panda.bin                             2020-02-05 11:47:44        13348
panda.bin                                      2020-02-05 11:47:44        34088
user1.bin                                      2020-02-05 11:47:46       248212
user2.bin                                      2020-02-05 11:47:46       248212
boot_v1.5.bin                                  2016-05-20 16:53:08         3232
version                                        2020-02-05 11:47:48           27
0. Preparing to flash b'v1.7.3-DEV-d034f3e9-RELEASE'
1. Entering DFU mode
opening device 41004c000251373034353037 0xddcc
connected
LIBUSB_ERROR_IO [-1]
2. Programming bootstub
programming 0 with length 2048
programming 1 with length 2048
programming 2 with length 2048
programming 3 with length 2048
programming 4 with length 2048
programming 5 with length 2048
programming 6 with length 2048
3. Flashing main code
opening device 41004c000251373034353037 0xddee
connected
flash: main version is v1.7.3-DEV-d034f3e9-RELEASE
flash: bootstub version is v1.7.3-DEV-d034f3e9-RELEASE
flash: unlocking
flash: erasing
flash: flashing
flash: resetting
reconnecting is taking 1 seconds...
opening device 41004c000251373034353037 0xddcc
connected
4. Flashing ESP (slow!)
opening device 41004c000251373034353037 0xddcc
connected
set baud to 230400
Connecting...
Connection timeout.
Connection timeout.
Connection timeout.
Connection timeout.
Connection timeout.
Connection timeout.
Connection timeout.
Connection timeout.
Connection timeout.
Connection timeout.
Connection timeout.
Connection timeout.
Connection timeout.
Connection timeout.
Connection timeout.
Connection timeout.

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/oguz/.local/lib/python3.7/site-packages/panda/flash_release.py", line 71, in flash_release
    esp.connect()
  File "/home/oguz/.local/lib/python3.7/site-packages/panda/esptool.py", line 195, in connect
    raise FatalError('Failed to connect to ESP8266')
panda.esptool.FatalError: Failed to connect to ESP8266

```

ozzdemir, 2020-04-20:
My original problem similar to #420 I think.

robbederks, 2020-04-21:
The `flash_release.py` script wasn't up to date for the different panda versions we have now. Only white pandas contain the ESP wifi chipset, so that step does not make sense on your black panda.

Usually, people flash their pandas by running `make` in the `panda/board` folder, or `make recover` if that doesn't work. Anyways, I fixed the `flash_release.py` too: #514 

As for your initial problem: please check that your panda is in a safety mode which allows for sending on the CAN buses. In the default safety mode, the panda acts read-only on the CAN bus. 
To allow all transmissions (including ACKing), you can try the ALLOUTPUT mode.

closed 2020-04-21 by robbederks