commaai/panda #537: Is this possible to be always true ?

deanlee, 2020-05-15:
https://github.com/commaai/panda/blob/94c995f0a9ae420124060cb406b44814dd80c82d/board/board.h#L26

robbederks, 2020-05-15:
Hi, are you sure this commit broke the panda for you? 
To test, you could run the `tests/debug_console.py` after rebooting the panda, and see what type of board it reports. If you're using a Comma 2, it should report as `Board type: Uno`

deanlee, 2020-05-16:
I am using the comma 1.previous generation device. 

> ************************ MAIN START ************************
Config:
  Board type: Black
  USB serial
  No bootmode
detected car harness with orientation 01
switching harness to passthrough (relay off)
setting CAN2 to be normal
switching harness to passthrough (relay off)
**** INTERRUPTS ON ****
ESUSP detected
USB reset
enumeration done
USB reset
enumeration done
USB enumeration complete
EON hasn't sent a heartbeat for 0x00000002 seconds. Safety is set to SILENT mode.
enable power savings
EON hasn't sent a heartbeat for 0x00000003 seconds. Safety is set to SILENT mode.
EON hasn't sent a heartbeat for 0x00000004 seconds. Safety is set to SILENT mode.
EON hasn't sent a heartbeat for 0x00000005 seconds. Safety is set to SILENT mode.
EON hasn't sent a heartbeat for 0x00000006 seconds. Safety is set to SILENT mode.
EON hasn't sent a heartbeat for 0x00000007 seconds. Safety is set to SILENT mode.
EON hasn't sent a heartbeat for 0x00000008 seconds. Safety is set to SILENT mode.
EON hasEON hasn't sent a heartbeat for 0x00000017 seconds. Safety is set to SILENT mode.

currently,I have ot comment out the if(!detect_with_pull(GPIOB, 1, PULL_UP)) statement to make it work again.

pd0wm, 2020-05-28:
@deanlee you mean a NEO board? https://github.com/commaai/neo

deanlee, 2020-05-29:
yes

robbederks, 2020-05-29:
Oh, I see. This was only tested on the new(er) generation pandas (white, grey, black, uno, ...), not on the NEO board. That board has been deprecated a very long time ago, actually surprised it still worked at all.
I see two solutions:
- Run a fork which maintains compatibility for the NEO
- Make a PR with another board for NEO, and figure out a way to detect this. If this is clean and doesn't break any functionality, I think we should be able to merge that

closed 2020-06-02 by robbederks