commaai/panda #546: USB OTEPDIS interrupt Error

xx979xx, 2020-05-29:
**when trying debug_console.py, I and others are getting this error continuously:
`OUTEP3 error 00000010`**
[image]

it is generated from [board/drivers/usb.h line 836](https://github.com/commaai/panda/blob/a618e64d1a1778f5b3ac95dcd8a41b64d1a18a3f/board/drivers/usb.h#L836) after receiving `USB_OTG_DOEPINT_OTEPDIS` interrupt which received when OUT token received when endpoint disabled.
the stm32f413 reference manual page 1220 added that: it "Applies only to control OUT endpoints.
Indicates that an OUT token was received when the endpoint was not yet enabled. This
interrupt is asserted on the endpoint for which the OUT token was received."

I am not sure if this a bug or happens by design so I can remove it from being printed as an error

robbederks, 2020-06-02:
Not sure why this triggers, have not seen this myself yet.
Do you have an easy way to reproduce?

xx979xx, 2020-06-08:
> Not sure why this triggers, have not seen this myself yet.
> Do you have an easy way to reproduce?
you can use `HKG_community` fork from [here](https://github.com/xx979xx/openpilot) as well as '7.6-dev`

robbederks, 2020-06-08:
@gregjhogan this is the same issue you reported it seems like? Will look into it

gregjhogan, 2020-06-08:
@robbederks looks the same, but when I encountered the issue all TX would stop

closed 2020-09-07 by robbederks