commaai/panda #560: Nissan safety angle tx bug

dzid26, 2020-06-18:
I noticed [this ](https://github.com/commaai/panda/blob/5b1494514042721b63f71254700ddcf94dbbd9c7/board/safety/safety_nissan.h#L119)doesn't make sense:
```
float delta_angle_float;
      delta_angle_float = (interpolate(NISSAN_LOOKUP_ANGLE_RATE_UP, nissan_speed) * NISSAN_DEG_TO_CAN) + 1.;
      int delta_angle_up = (int)(delta_angle_float);
      delta_angle_float =  (interpolate(NISSAN_LOOKUP_ANGLE_RATE_DOWN, nissan_speed) * NISSAN_DEG_TO_CAN) + 1.;
      int delta_angle_down = (int)(delta_angle_float);
```      

I think it is supposed to be:
```float delta_angle_float;
      delta_angle_float = (int)(delta_angle_float); 
      int delta_angle_up = (interpolate(NISSAN_LOOKUP_ANGLE_RATE_UP, nissan_speed) * NISSAN_DEG_TO_CAN) + 1.;
      int delta_angle_down  =  (interpolate(NISSAN_LOOKUP_ANGLE_RATE_DOWN, nissan_speed) * NISSAN_DEG_TO_CAN) + 1.;
```

robbederks, 2020-06-18:
Not sure what you mean is wrong. Looks correct to me?

dzid26, 2020-06-18:
The `violation |= max_limit_check(desired_angle, highest_desired_angle, lowest_desired_angle);
` will always be `true`.

You've got the same thing to different variables:
```
int delta_angle_up      = (int)(delta_angle_float);
int delta_angle_down = (int)(delta_angle_float);
```
And then different things assigned to the same variable:
```
delta_angle_float = (interpolate(NISSAN_LOOKUP_ANGLE_RATE_UP, vehicle_speed) * NISSAN_DEG_TO_CAN) + 1.;
delta_angle_float = (interpolate(NISSAN_LOOKUP_ANGLE_RATE_DOWN, vehicle_speed) * NISSAN_DEG_TO_CAN) + 1.;
```

jyoung8607, 2020-06-18:
Look at the order carefully. If it helps, here's different pseudocode:

```c
float temp_scratchpad;

temp_scratchpad = floating point version of value 1;
int value1 = convert temp_scratchpad to int;

temp_scratchpad = floating point version of value 2;
int value2 = convert temp_scratchpad to int;

do_stuff_that_needs_ints(value1, value2);
```
I didn't write the Nissan safety code and haven't looked much beyond your question for the rationale here, but I can tell you [MISRA](https://en.wikipedia.org/wiki/MISRA_C) is very picky about types and casting, and this style might have been the most expedient way to meet the guidelines.

dzid26, 2020-06-19:
Ok, I was looking at it the wrong way for whatever reason! Thanks

Strange that the inline casting wasn't compliant.

closed 2020-06-19 by dzid26