commaai/panda #569: Can't change safety mode to ALLOUTPUT

omertal323, 2020-07-08:
p.change_safety_mode(17) is not changing safety mode (according to p.health()). every other safety mode works. This issue blocks me from sending arbitrary CAN messages.

jyoung8607, 2020-07-08:
Be sure you have flashed Panda with debug firmware, otherwise certain community-supported safety modes and the all-output safety mode aren't enabled. If testing with something other than openpilot, you'll also need the EON option unset, or the lack of boardd keepalives will set Panda back to no output a few seconds later. Try building from source and flashing on Ubuntu, where I believe both of those will happen by default.

closed 2020-07-08 by jyoung8607

omertal323, 2020-07-09:
Thanks for the help

I can't flash anything, the LED is flashing green and device stop
responding till power reset...

omertal323, 2020-07-09:
I seemd to have solved the problem! there was a problem with the udev rule
in the readme, the syntax is not corrent to my ubuntu machine. Anyway, now
i managed to flashed a debug version and successfully change into ALLOUTPUT
safety mode.

If I may ask another question, is there a way to enable this mode on WIFI
too? this is really important to me since this is the reason I purchased
the device at all :)

Thanks,
Omer