commaai/panda #576: USB controlRead call that uses Panda.REQUEST_OUT

gregjhogan, 2020-07-23:
python `libusb` wrapper ignores `REQUEST_IN`/`REQUEST_OUT` when you call `controlRead()`/`controlWrite()`
https://github.com/vpelletier/python-libusb1/blob/697b6cddb91c87da17d94cdc3b0733b260ee0e7a/usb1/__init__.py#L1506
https://github.com/vpelletier/python-libusb1/blob/697b6cddb91c87da17d94cdc3b0733b260ee0e7a/usb1/__init__.py#L1486

so this works but is very confusing (`REQUEST_IN` direction is actually in use):
https://github.com/commaai/panda/blob/52583dc289a4371e3647d06ce7f91af2ec9d4820/python/__init__.py#L602-L609

since the name implies the direction, the direction really shouldn't be specified (the param exists for other flags)

pd0wm, 2020-07-24:
The flags are also ignored on the panda side

adeebshihadeh, 2023-08-08:
Not sure what this issue is for? Is it still relevant?

closed 2024-02-21 by adeebshihadeh