commaai/panda #603: Panda Logger

TK211X, 2020-10-10:
When Panda detects a flash drive or removable storage media with no EON/C2 connected it should start to write CAN logs to it until it's full, thence overwrite oldest data. 

CAN logs should be time stamped from vehicle start or start of log along with bus number.

This will allow discrete logging of data in sensitive situations where having a EON/C2 is inappropriate or not feasible. It will also allow for unique opportunities to BUG / wire Panda into remote spots and be stowed out of sight.

And example would be to log dealer activity on OBD by tapping the rear of an OBD and stowing the Panda out of sight.

[image]

This can also be done with a convectional harness setup with a black panda. Simply plug the USB C into the black panda and remove C2. Insert flash drive and replace cover. Start getting logs. The bonus here is that the camera BUS and IDs will be logged and some users won't need to purchase and return a C2 to find out if their vehicle can be supported. Reducing the workload for returns on comma employees.

Many benefits. One feature. Especially since Chffr is gone and removed the option to get wireless data logs and using Cabana live isn't always and option.

This should for sure be a basic feature shipped on every Panda.

Not sure why it was never thought of before.

robbederks, 2020-10-19:
Implementing this is not trivial (you'd need the STM to be in USB host mode, needs to know about filesystems, ...), and imo it's not useful enough to warrant that amount of dev effort.
This is ideal for an open source contribution on a fork though :)

TK211X, 2020-10-19:
You agree it’s not trivial, yet it would be a key feature of any Panda shipped. Adding value to a product that Comma recovers profit on through sales and has the potential to secure data for future developmental return for free through users leveraging the feature.

It’s not difficult to see without this and with the loss of EON support that the majority of users would have less of a reason to purchase black panda on its own. You can’t deny the majority of purchases come from relatively unskilled users expecting capability without planning on making a significant contribution.

TK211X, 2020-10-19:
@geohot Add key features without waiting for ages for someone to do it for free. 

[image]

Expedite growth; bro.

geohot, 2020-10-19:
The future of comma is not the panda, it's the comma two and products like that. When panda run out, we aren't going to make more.

panda is open source, feel free to write this. In fact, fork the panda code, write this closed source, buy panda from us, install custom feature that will make buku bucks, lock firmware, resell panda for buku more bucks, and enjoy being rich.

closed 2020-10-19 by geohot