commaai/panda #610: Can't Access Panda

BenderV, 2020-11-03:
Hello,

I'm having trouble accessing the (White) Panda.

Using the [get_panda_password](https://github.com/commaai/panda/blob/master/examples/get_panda_password.py):
```
Trying to connect to Panda over USB...
USB connection failed
```

The Panda display a blue dot and blink a red dot.

I'm quite lost to where should I begin debugging that, and would love some help 👍 

Thanks !

amandeepssk, 2020-11-09:
I am also facing exactly the same issue. An update on this issue would be a great help.

adeebshihadeh, 2020-12-18:
The wifi functionality of white pandas is deprecated, however see the usage section of the README for help connecting to the panda - https://github.com/commaai/panda#usage.

closed 2020-12-18 by adeebshihadeh

BenderV, 2020-12-18:
Hi @adeebshihadeh,
I am sorry, but I don't see how you comment is relevant here. I'm not talking about Panda. I can't connect over USB.

BenderV, 2020-12-31:
Hi @adeebshihadeh,
Can you please re-open this issue ?
Thanks