commaai/panda #621: IR LEDs constantly on in GM Volt

VirtuallyChris, 2021-01-21:
Volt people have a different ignition detection that detects CAN messages on the bus instead of detecting a rise in the ignition line. Due to this, the IR LEDs never turn off. (at least is what it seems like!) Confirmed with several GM users on Discord that this happens to them, and they're forced to unplug their devices after each drive.
[image]

robbederks, 2021-01-27:
I briefly looked into this a few weeks ago, and none of the logic about when to turn the IR LEDs on/off is in the panda (except for the heartbeat being missing turning them off).
Also looked into the openpilot side, but there the remainder of the code seems to be shared between both ignition types when the health packet comes in. 
Requires more investigation!

dibble9012, 2021-01-28:
Happens to me
Device Dongle ID: 970dccbaef8569a7
2017 Chevy Volt

ErikRNissen, 2021-01-29:
I’m seeing this on a2018 volt as well.   Any time I check the interior cam the ir LEDs appear to be on 
[image]

Verylukyguy, 2021-01-30:
Happens to me.

Dongle ID:  e38a02cae1ef1e0c

2018 GMC Acadia Denali AWD

jeffpicard, 2021-02-01:
+1 
Dongle ID: 9b7062f1d55b8e78
2017 Chevy Volt

My comma2 shuts off if i haven't driven the car for around 2 days. Ditto ErikRNissen -- my interior cam always shows the IR lights on.

robbederks, 2021-02-01:
Thanks for the reports, we're going to look into this issue further.

As far as the IR being on while taking a snapshot, that's completely expected behavior and not indicative of any issues.
The problem this issue is about is the IR being on while the device is offroad / idle. When taking a snapshot, the IR LEDs turn on for a brief duration.

robbederks, 2021-02-01:
Dug a little deeper and found the issue. The issue is indeed related to the panda code.

The IR PWM controller uses timer 4 (TIM4), which apparently is also used by the GMLAN bitbanging driver.
Will move the GMLAN bitbanging to a different timer to fix this issue. PR coming soon!

dibble9012, 2021-02-01:
Awesome!!! Now I don’t have to worry about a dead battery after three days. Thank you!

PR #627 (2021-02-01): Move gmlan_alt to TIM12 to fix concurrency issue with IR PWM
To fix #621
closed 2021-02-01 by robbederks