commaai/panda #722: panda linux driver build fail on Ubuntu 20.04.3 and 21.04

hortovanyi, 2021-09-16:
run `make all` under pans/drivers/linux
```
sudo dkms build panda/0.0.1
[sudo] password for nick: 

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.11.0-34-generic -C /lib/modules/5.11.0-34-generic/build M=/var/lib/dkms/panda/0.0.1/build...(bad exit status: 2)
ERROR (dkms apport): binary package for panda: 0.0.1 not found
Error! Bad return status for module build on kernel: 5.11.0-34-generic (x86_64)
Consult /var/lib/dkms/panda/0.0.1/build/make.log for more information.
make: *** [Makefile:8: build] Error 10
```

/var/lib/dkms/panda/0.0.1/build/make.log contains
```
DKMS make.log for panda-0.0.1 for kernel 5.11.0-34-generic (x86_64)
Thu 16 Sep 2021 14:42:21 AEST
make: Entering directory '/usr/src/linux-headers-5.11.0-34-generic'
  CC [M]  /var/lib/dkms/panda/0.0.1/build/panda.o
/var/lib/dkms/panda/0.0.1/build/panda.c: In function ‘panda_usb_process_can_rx’:
/var/lib/dkms/panda/0.0.1/build/panda.c:286:17: error: implicit declaration of function ‘get_can_dlc’ [-Werror=implicit-function-declaration]
  286 |   cf->can_dlc = get_can_dlc(msg->bus_dat_len & PANDA_DLC_MASK);
      |                 ^~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:288: /var/lib/dkms/panda/0.0.1/build/panda.o] Error 1
make: *** [Makefile:1848: /var/lib/dkms/panda/0.0.1/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.11.0-34-generic'
```
Not sure how to fix? get_can_dlc is a #define .. git pulled the latest code

adeebshihadeh, 2021-09-16:
This is being fixed in #700.

hortovanyi, 2021-09-16:
@adeebshihadeh thanks. Manually tried it on both versions of Ubuntu and now the driver installs

adeebshihadeh, 2021-11-04:
Fixed by bb5e2ccd0030052f5f9fd2e213ddb55864becc66

closed 2021-11-04 by adeebshihadeh