commaai/panda #771: Angle control safety: make sure requested angle is close to current angle

robbederks, 2021-11-09:


adeebshihadeh, 2024-02-21:
@sshane this still an issue?

sshane, 2024-02-21:
@robbederks what was the thinking on this? Was this to limit torque when overriding? For Nissan, we can directly control max torque, for Toyota we wind down steering on steer pressed, and with Ford, it has logic to limit angle near desired for overriding.

robbederks, 2024-02-21:
Can't remember what the thought process behind this was. I know the EPS will fault on Tesla if you request an angle too far away (effectively rate limiting it), but we already do this iirc.

Unless something else pops in mind, I think we can close this

sshane, 2024-02-21:
Sounds like this should be some fault avoidance code in openpilot if that is the issue

closed 2024-02-21 by sshane