commaai/panda #823: Look into slow safety tests

adeebshihadeh, 2022-01-12:
The safety tests got significantly slower in the last few months.

jyoung8607, 2022-01-12:
Hacking on car ports from time to time, I've noticed there are condition where sometimes they run fast and sometimes they run much slower to get the same apparent result.

I *think* there's a correlation to whether the Panda tree is checked out inside an openpilot tree or if it's been checked out standalone. The openpilot tree may need to be built. I can't prove that right now and I never tried to run down why.

adeebshihadeh, 2022-01-12:
I also suspect it has something to do with openpilot. I started to `git bisect`, but commits from 6 months ago are still super slow.

jyoung8607, 2022-01-12:
Part of #793 massively changed the cost of `test_fwd_hook()` and `test_spam_can_buses()`.

Current master:
```
((openpilot) ) jyoung@DESKTOP-6JPRDTA:~/openpilot/panda/tests/safety$ time HW_TYPE=0 python3 -m unittest discover .
.................................................................................................................s..................................................................s...................................................................................................................................................................................................................................................sssssssssssssssssss...................sssssssssssssssss.................................................................................s..................................................
----------------------------------------------------------------------
Ran 608 tests in 194.031s

OK (skipped=39)

real    3m14.340s
user    3m14.289s
sys     0m0.260s
```

With part of #793 backed out:

```diff
diff --git a/tests/safety/common.py b/tests/safety/common.py
index 8f11bb4..8bbe049 100644
--- a/tests/safety/common.py
+++ b/tests/safety/common.py
@@ -293,7 +293,7 @@ class PandaSafetyTest(PandaSafetyTestBase):
   def test_fwd_hook(self):
     # some safety modes don't forward anything, while others blacklist msgs
     for bus in range(0x0, 0x3):
-      for addr in range(0x1, 0x40000):
+      for addr in range(0x1, 0x800):
         # assume len 8
         msg = make_msg(bus, addr, 8)
         fwd_bus = self.FWD_BUS_LOOKUP.get(bus, -1)
@@ -302,7 +302,7 @@ class PandaSafetyTest(PandaSafetyTestBase):
         self.assertEqual(fwd_bus, self.safety.safety_fwd_hook(bus, msg), f"{addr=:#x} from {bus=} to {fwd_bus=}")

   def test_spam_can_buses(self):
-    for addr in range(1, 0x40000):
+    for addr in range(1, 0x800):
       for bus in range(0, 4):
         if all(addr != m[0] or bus != m[1] for m in self.TX_MSGS):
           self.assertFalse(self._tx(make_msg(bus, addr, 8)))
```

```
((openpilot) ) jyoung@DESKTOP-6JPRDTA:~/openpilot/panda/tests/safety$ time HW_TYPE=0 python3 -m unittest discover .
.................................................................................................................s..................................................................s...................................................................................................................................................................................................................................................sssssssssssssssssss...................sssssssssssssssss.................................................................................s..................................................
----------------------------------------------------------------------
Ran 608 tests in 2.869s

OK (skipped=39)

real    0m3.176s
user    0m3.189s
sys     0m0.207s
```

The PR landed on December 5th. My last big push on `mqb-long` was right about that time, and I had a couple of working trees going. I think I was doing occasional merges with master and I probably had trees on either side of that commit.

I realize why you did that, to scan the full 29 bit address range. Unsure at a glance if it could become much cheaper to run.

PR #827 (2022-01-26): Better CAN address range scanning
* Smarter coverage ranges for `test_relay_malfunction`, `test_fwd_hook`, and `test_spam_can_buses`
* Ensure that any new TX_MSGS outside the coverage range will fail testing until the range is expanded
* Minor consistency cleanup

Resolves #823
closed 2022-01-26 by adeebshihadeh