commaai/panda #865: Linux driver does not compile with 5.6.11 kernel

2m, 2022-03-01:
I get the following error:

```
DKMS make.log for panda-0.0.1 for kernel 5.16.11-arch1-1 (x86_64)
Tue Mar  1 08:54:30 PM EET 2022
make: Entering directory '/usr/lib/modules/5.16.11-arch1-1/build'
  CC [M]  /var/lib/dkms/panda/0.0.1/build/panda.o
/var/lib/dkms/panda/0.0.1/build/panda.c: In function ‘panda_usb_write_bulk_callback’:
/var/lib/dkms/panda/0.0.1/build/panda.c:185:3: error: too few arguments to function ‘can_get_echo_skb’
  185 |   can_get_echo_skb(netdev, ctx->ndx);
      |   ^~~~~~~~~~~~~~~~
In file included from ./include/linux/can/dev.h:23,
                 from /var/lib/dkms/panda/0.0.1/build/panda.c:13:
./include/linux/can/skb.h:24:14: note: declared here
   24 | unsigned int can_get_echo_skb(struct net_device *dev, unsigned int idx,
      |              ^~~~~~~~~~~~~~~~
/var/lib/dkms/panda/0.0.1/build/panda.c: In function ‘panda_usb_start_xmit’:
/var/lib/dkms/panda/0.0.1/build/panda.c:462:3: error: too few arguments to function ‘can_put_echo_skb’
  462 |   can_put_echo_skb(skb, priv_inf->netdev, ctx->ndx);
      |   ^~~~~~~~~~~~~~~~
In file included from ./include/linux/can/dev.h:23,
                 from /var/lib/dkms/panda/0.0.1/build/panda.c:13:
./include/linux/can/skb.h:20:5: note: declared here
   20 | int can_put_echo_skb(struct sk_buff *skb, struct net_device *dev,
      |     ^~~~~~~~~~~~~~~~
/var/lib/dkms/panda/0.0.1/build/panda.c:487:3: error: too few arguments to function ‘can_free_echo_skb’
  487 |   can_free_echo_skb(priv_inf->netdev, ctx->ndx);
      |   ^~~~~~~~~~~~~~~~~
In file included from ./include/linux/can/dev.h:23,
                 from /var/lib/dkms/panda/0.0.1/build/panda.c:13:
./include/linux/can/skb.h:26:6: note: declared here
   26 | void can_free_echo_skb(struct net_device *dev, unsigned int idx,
      |      ^~~~~~~~~~~~~~~~~
make[1]: *** [scripts/Makefile.build:287: /var/lib/dkms/panda/0.0.1/build/panda.o] Error 1
make: *** [Makefile:1846: /var/lib/dkms/panda/0.0.1/build] Error 2
make: Leaving directory '/usr/lib/modules/5.16.11-arch1-1/build'

 ```

groygirard, 2022-03-16:
Same issue under Ubuntu 20.04 kernel 5.13.0-35
```
Building module:
cleaning build area...
make -j12 KERNELRELEASE=5.13.0-35-generic -C /lib/modules/5.13.0-35-generic/build M=/var/lib/dkms/panda/0.0.1/build...(bad exit status: 2)
ERROR (dkms apport): binary package for panda: 0.0.1 not found
Error! Bad return status for module build on kernel: 5.13.0-35-generic (x86_64)
Consult /var/lib/dkms/panda/0.0.1/build/make.log for more information.
make: *** [Makefile:8: build] Error 10
```
```
DKMS make.log for panda-0.0.1 for kernel 5.13.0-35-generic (x86_64)
Wed 16 Mar 2022 05:37:30 PM EDT
make: Entering directory '/usr/src/linux-headers-5.13.0-35-generic'
  CC [M]  /var/lib/dkms/panda/0.0.1/build/panda.o
/var/lib/dkms/panda/0.0.1/build/panda.c: In function ‘panda_usb_write_bulk_callback’:
/var/lib/dkms/panda/0.0.1/build/panda.c:185:3: error: too few arguments to function ‘can_get_echo_skb’
  185 |   can_get_echo_skb(netdev, ctx->ndx);
      |   ^~~~~~~~~~~~~~~~
In file included from ./include/linux/can/dev.h:23,
                 from /var/lib/dkms/panda/0.0.1/build/panda.c:13:
./include/linux/can/skb.h:24:14: note: declared here
   24 | unsigned int can_get_echo_skb(struct net_device *dev, unsigned int idx,
      |              ^~~~~~~~~~~~~~~~
/var/lib/dkms/panda/0.0.1/build/panda.c: In function ‘panda_usb_start_xmit’:
/var/lib/dkms/panda/0.0.1/build/panda.c:462:3: error: too few arguments to function ‘can_put_echo_skb’
  462 |   can_put_echo_skb(skb, priv_inf->netdev, ctx->ndx);
      |   ^~~~~~~~~~~~~~~~
In file included from ./include/linux/can/dev.h:23,
                 from /var/lib/dkms/panda/0.0.1/build/panda.c:13:
./include/linux/can/skb.h:20:5: note: declared here
   20 | int can_put_echo_skb(struct sk_buff *skb, struct net_device *dev,
      |     ^~~~~~~~~~~~~~~~
/var/lib/dkms/panda/0.0.1/build/panda.c:487:3: error: too few arguments to function ‘can_free_echo_skb’
  487 |   can_free_echo_skb(priv_inf->netdev, ctx->ndx);
      |   ^~~~~~~~~~~~~~~~~
In file included from ./include/linux/can/dev.h:23,
                 from /var/lib/dkms/panda/0.0.1/build/panda.c:13:
./include/linux/can/skb.h:26:6: note: declared here
   26 | void can_free_echo_skb(struct net_device *dev, unsigned int idx,
      |      ^~~~~~~~~~~~~~~~~
make[1]: *** [scripts/Makefile.build:281: /var/lib/dkms/panda/0.0.1/build/panda.o] Error 1
make: *** [Makefile:1879: /var/lib/dkms/panda/0.0.1/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.13.0-35-generic
```

adeebshihadeh, 2022-04-05:
Fixed with ddf1eb68ba6324022b95103994b11bedb615f608

closed 2022-04-05 by adeebshihadeh