commaai/panda #968: can't transmit from USB if any queue is full

jyoung8607, 2022-06-16:
Had an interesting case on a new car port, where openpilot's sent messages started being blackholed almost immediately after startup. Traced it back to the bus 2 side not being hooked up while Panda was attempting bus 0->2 forwarding as it normally would. After locally disabling forwarding because we didn't need it on this particular car, everything started working properly.

Before: `578742b26807f756|2022-06-14--15-38-12`
After: `578742b26807f756|2022-06-14--20-59-29`

Symptom is that only the first ~250ms of messages made it out, everything else was blackholed. openpilot logs revealed lots of:

`[176.782847] selfdrive/boardd/panda.cc:213 - usb_bulk_write: Transmit buffer full`

Apparently driven by the fact USB bulk transfers don't get completed ...

https://github.com/commaai/panda/blob/6c1738814b451d23797ab7d8fddd43c3b5296b45/board/drivers/bxcan.h#L146-L158

... until a check for > 51 of 416 available transmit ring entries passes ...

https://github.com/commaai/panda/blob/6c1738814b451d23797ab7d8fddd43c3b5296b45/board/usb_comms.h#L68-L72

... on *all* transmit queues.

https://github.com/commaai/panda/blob/6c1738814b451d23797ab7d8fddd43c3b5296b45/board/drivers/can_common.h#L218-L224

And the Panda-side forwarding from 0->2 with no CAN acknowledgements on bus 2 kept that transmit queue too full for USB to get a packet in edgewise.

Something does have to be done if queues fill up, but that can be head-of-line or tail-of-line discarding, rather than globally stalling USB transfers with resulting USB stack confusion, and possible disruption with boardd's realtime ratekeeping.

This has a potential nexus with commaai/openpilot#24475. We still need better recovery logic, but this could explain how a steering message got silently discarded (as was discovered analyzing the logs for that incident), and perhaps why it's only happening to one vehicle (because that vehicle might have physical trouble resulting in a CAN transmit queue filling up).

gregjhogan, 2022-06-16:
This was by design to support an operating mode with guaranteed delivery of all messages sent to the panda (for situations like flashing firmware).  Once the panda receives a USB packet it has to process it (you can't reject it after it comes in) so we have to block incoming USB packets unless all TX buffers have available space for this mode of operation.  The way you block future USB packets is by not clearing the NAK on the bulk TX endpoint, so we only do that when all TX buffers have available space.  I am not sure what the solution is, but we need to support the current behavior, too.

jyoung8607, 2022-06-16:
> This was by design to support an operating mode with guaranteed delivery of all messages sent to the panda (for situations like flashing firmware). ... I am not sure what the solution is, but we need to support the current behavior, too.

Fair and reasonable... possibilities:

* Switch to Panda-side discarding when in a vehicle safety mode, so stalls on one bus don't propagate to others
* Use a different USB bulk endpoint for each CAN bus to get independent NAK flow control

adeebshihadeh, 2023-08-06:
This currently works as intended. Using a different endpoint for each CAN TX buffer should be a strict improvement, but unlikely we'll get around to it unless this starts affecting openpilot use. Would definitely merge a PR for it though.

closed 2023-08-06 by adeebshihadeh