commaai/panda #984: Docker build fails "No module named 'pkgconfig'"

series8217, 2022-07-09:
Building Dockerfile.panda fails on the following step:

`Step 18/20 : RUN cd /tmp/openpilot &&     pip install --no-cache-dir -r opendbc/requirements.txt &&     pip install --no-cache-dir --upgrade aenum lru-dict pycurl tenacity atomicwrites serial smbus2 scons`

The error occurs while it is installing dependency pycapnp from opendbc/requirements.txt:

```ERROR: Command errored out with exit status 1:
     command: /root/.pyenv/versions/3.8.10/bin/python3.8 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-ny0gasxi/pycapnp_c1bd937f9799462ea4c55c569b6183fc/setup.py'"'"'; __file__='"'"'/tmp/pip-install-ny0gasxi/pycapnp_c1bd937f9799462ea4c55c569b6183fc/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-eabo_io2
         cwd: /tmp/pip-install-ny0gasxi/pycapnp_c1bd937f9799462ea4c55c569b6183fc/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-ny0gasxi/pycapnp_c1bd937f9799462ea4c55c569b6183fc/setup.py", line 12, in <module>
        import pkgconfig
    ModuleNotFoundError: No module named 'pkgconfig'
    ----------------------------------------
```

Docker version is `20.10.17, build 100c701`. Arch is arm.

Has anyone run into this?   I'll keep poking around tomorrow and post a solution or PR if I find the issue.

series8217, 2022-07-11:
This is actually an upstream issue with opendbc: https://github.com/commaai/opendbc/issues/663

adeebshihadeh, 2022-08-03:
Closing as this is a duplicate of the opendbc issue.

closed 2022-08-03 by adeebshihadeh