commaai/panda #1021: Red Panda "dead" after being connected to C3 wired for ev6 setup

razem-io, 2022-08-10:
Today, I tried to connect my red panda to c3 to get my ev6 to work with openpilot.

At first, everything looked okay. Red panda's led was blue when it wasn't connected to c3. As soon as a connection could be established, it was blinking green. During a reboot of the c3 it began to blink green fast (like a strobe light) and turned solid green for a few seconds once the boot process was finished. Then it blinked green / red (also very fast) for 1-2 seconds and completely turned silent.

My setup:
- red panda (purchased long time ago but was never used till today)
- c3
- op 0.8.15
- connected using the diagram pinned in the kia / genesis / hyundai discord channel for the ev6

What I tried:
- sshing into c3 and checking if I can see the device using lsusb. Sadly, no luck there.
- connecting it to my laptop, checking out the panda repo, installing everything needed from get-sdk (but on arch) and installed the linux dkms drivers. Also, no luck detecting the device using lsusb.

A few other users owning an ev6 on the discord server have the same problem. Ofc, I can not be sure if they are just in DFU mode.

What are my options now? Can I fix the device without sending it back over the ocean? Any idea what might have caused this?

adeebshihadeh, 2022-08-10:
What's your dongle ID? Are you on 0.8.15-release?

razem-io, 2022-08-10:
Dongle-ID: ba3e03ccb4b26329
Yes on 0.8.15. C3 received a hard reset to factory default yesterday.

adeebshihadeh, 2022-08-10:
Can you try running master or master-ci?

razem-io, 2022-08-10:
Sure, I will try that first thing tomorrow. Thanks! 

However, can this even work if lsusb does not show anything on the c3?

razem-io, 2022-08-11:
Switched to master. Sadly, red panda does not show up. Trying to run recover_h7.sh returns the following output (same if red panda is not connected):

```
comma@tici:/data/openpilot/panda/board$ git status
HEAD detached at 89989ab
comma@tici:/data/openpilot/panda/board$ ./recover_h7.sh
scons: Entering directory `/data/openpilot/panda'
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: `board' is up to date.
scons: done building targets.
opening device 33001f001951393037323631 0xddcc
exception LIBUSB_ERROR_BUSY [-6]
Traceback (most recent call last):
  File "/data/openpilot/panda/python/__init__.py", line 242, in connect
    self._handle.claimInterface(0)
  File "/usr/local/pyenv/versions/3.8.10/lib/python3.8/site-packages/usb1/__init__.py", line 1146, in claimInterface
    mayRaiseUSBError(
  File "/usr/local/pyenv/versions/3.8.10/lib/python3.8/site-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/usr/local/pyenv/versions/3.8.10/lib/python3.8/site-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorBusy: LIBUSB_ERROR_BUSY [-6]
connected
opening device 33001f001951393037323631 0xddee
connected
opening device 33001f001951393037323631 0xddee
connected
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
dfu-util: Cannot claim interface
```

razem-io, 2022-08-11:
GREAT... now it flashed the internal black panda because boardd wasn't running, and black panda does not show up as well 🤦.

Is it okay, if I update the wiki, to NOT execute `recover_h7.sh` (even if a red panda is connected) on a c3?
https://github.com/commaai/openpilot/wiki/Panda-Flashing 

Also, is there anyway to recover from this? Hopefully I haven't bricked a $2000 device just by doing that.

razem-io, 2022-08-11:
Alright, got my black panda back. I haven't found a wiki entry yet. Also, not sure, if this should be part of the wiki. That's why I document it here for now, since it is relevant for the issue.

You can ignore `-bash: echo: write error: Device or resource busy` while executing the commands below. 

To get black panda on c3 into DFU mode:
```
echo 124 > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio124/direction
echo 1 > /sys/class/gpio/gpio124/value
 
echo 134 > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio134/direction
echo 1 > /sys/class/gpio/gpio134/value
 
echo 124 > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio124/direction
echo 0 > /sys/class/gpio/gpio124/value
```

lsusb should now output `STMicroelectronics STM Device in DFU Mode`. You can now run recover.sh in the panda folder.

Get black panda on c3 out of DFU mode:
```
echo 124 > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio124/direction
echo 1 > /sys/class/gpio/gpio124/value
 
echo 134 > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio134/direction
echo 0 > /sys/class/gpio/gpio134/value
 
echo 124 > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio124/direction
echo 0 > /sys/class/gpio/gpio124/value
```

Black panda should be visible again by executing lsusb. E.g. `Bus 001 Device 006: ID bbaa:ddcc`.

adeebshihadeh, 2022-08-11:
I looked into this and found it's the result of two bugs: the missing red panda firmware file on the release branch (https://github.com/commaai/openpilot/pull/25401) and the broken bootstub flashing (#1007). I'm pushing a hotfix to the release to address this. Unfortunately, it looks like you flashed the F4 firmware to your red panda, which is why it doesn't show up over USB . A [panda paw](https://github.com/commaai/neo/tree/master/paw) will allow you to flash the correct firmware.o/tree/master/paw will allow you to flash the correct firmware on your red panda.

> Also, is there anyway to recover from this? Hopefully I haven't bricked a $2000 device just by doing that.

Looks like you've already recovered your comma three's panda. The comma three essentially has a built-in panda paw, so it's pretty unbrickable.

Thanks for reporting the bug! We've already written tests to cover one of the bugs and plan to build a full test suite to comprehensively cover the flashing procedure.

closed 2022-08-11 by adeebshihadeh

razem-io, 2022-08-11:
Ordered a "panda paw" today. Thank you for the quick hotfix!