commaai/panda #1202: SAFETY_ALLOUTPUT and can_send overwhelm the car with white panda

lvlas, 2023-01-04:
I have this simple code in my car running on white panda. It works but I can do almost nothing in the car after it, seems like the can bus is overwhelmed. I have to unplug the panda or write panda.reset() to even be able to turn the radio on.  Any way to solve this please? I have no problem with these commands using socketcan and different obd hardware

panda = Panda()
panda.set_safety_mode(Panda.SAFETY_ALLOUTPUT)
panda.can_send(0x01a, b'\x11\x22\x33\x44\xAA\xBB\xCC\xDD', 0)
sleep(1)
panda.can_send(0x1c0, b'\x24\x74\x6C\x90\x11\x00', 0)            
panda.close()

dzid26, 2023-03-22:
maybe check if you are sending with the right rate, eg:
`panda.set_can_speed_kbps(0, 500)`

adeebshihadeh, 2023-06-17:
Sounds like the panda is either retransmitting the message or sending error frames. You can get more info to debug with `Panda.health` and ` Panda.can_health`.

closed 2023-06-17 by adeebshihadeh